Ultimaker/Cura #2045: Cura Crash Report

AB-Normal1960, 2017-06-29:
I am running Cura. I downloaded a Sketchup file and used the free Sketchup Maker program to export it to a .obj file I thought Cura could use.  When I tried to open it in Cura, an error file was generated and it asked me to send it to you all as a bug report.  Here is the data from the generated error screen:
Version: 2.6.0
Platform: Windows-10-10.0.15063
Qt: 5.7.1
PyQt: 5.7.1

Exception:
Traceback (most recent call last):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 685, in event
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 261, in event
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Event.py", line 189, in call
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 315, in __performEmit
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 1291, in _readMeshFinished
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\Arrange.py", line 56, in create
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\ShapeArray.py", line 33, in fromPolygon
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\ShapeArray.py", line 70, in arrayFromPolygon
MemoryError

-------------------------
Thread 0x0000139c (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 150 in _storeStderrToLogThread
  File "D:\env\2.6\inst\Lib\threading.py", line 862 in run
  File "D:\env\2.6\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.6\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000044 (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Backend\Backend.py", line 142 in _storeOutputToLogThread
  File "D:\env\2.6\inst\Lib\threading.py", line 862 in run
  File "D:\env\2.6\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.6\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002d44 (most recent call first):
  File "C:\Program Files\Cura 2.6\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "D:\env\2.6\inst\Lib\threading.py", line 862 in run
  File "D:\env\2.6\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.6\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003508 (most recent call first):
  File "D:\env\2.6\inst\Lib\threading.py", line 293 in wait
  File "D:\env\2.6\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.6\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.6\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001c08 (most recent call first):
  File "D:\env\2.6\inst\Lib\threading.py", line 293 in wait
  File "D:\env\2.6\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.6\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.6\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x000026d8 (most recent call first):
  File "D:\env\2.6\inst\Lib\threading.py", line 293 in wait
  File "D:\env\2.6\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.6\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.6\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003150 (most recent call first):
  File "D:\env\2.6\inst\Lib\threading.py", line 293 in wait
  File "D:\env\2.6\inst\Lib\threading.py", line 424 in acquire
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 86 in _nextJob
  File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\JobQueue.py", line 116 in run
  File "D:\env\2.6\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.6\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003424 (most recent call first):
  File "D:\env\2.6\inst\Lib\threading.py", line 297 in wait
  File "D:\env\2.6\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.6\inst\lib\site-packages\zeroconf.py", line 1153 in run
  File "D:\env\2.6\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.6\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00002f08 (most recent call first):
  File "D:\env\2.6\inst\Lib\threading.py", line 297 in wait
  File "D:\env\2.6\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.6\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.6\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.6\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001c74 (most recent call first):
  File "D:\env\2.6\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.6\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.6\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x0000335c (most recent call first):
  File "C:\Program Files\Cura 2.6\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 58 in _updateThread
  File "D:\env\2.6\inst\Lib\threading.py", line 862 in run
  File "D:\env\2.6\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.6\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000bd8 (most recent call first):
  File "D:\env\2.6\inst\Lib\threading.py", line 297 in wait
  File "D:\env\2.6\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.6\inst\lib\site-packages\zeroconf.py", line 1153 in run
  File "D:\env\2.6\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.6\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x000016f0 (most recent call first):
  File "D:\env\2.6\inst\Lib\threading.py", line 297 in wait
  File "D:\env\2.6\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.6\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.6\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.6\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00003700 (most recent call first):
  File "D:\env\2.6\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.6\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.6\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00001fd8 (most recent call first):
  File "D:\env\2.6\inst\Lib\threading.py", line 297 in wait
  File "D:\env\2.6\inst\lib\site-packages\zeroconf.py", line 1569 in wait
  File "D:\env\2.6\inst\lib\site-packages\zeroconf.py", line 1017 in run
  File "D:\env\2.6\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.6\inst\Lib\threading.py", line 882 in _bootstrap

Thread 0x00000cbc (most recent call first):
  File "D:\env\2.6\inst\lib\site-packages\zeroconf.py", line 946 in run
  File "D:\env\2.6\inst\Lib\threading.py", line 914 in _bootstrap_inner
  File "D:\env\2.6\inst\Lib\threading.py", line 882 in _bootstrap

Current thread 0x00001de8 (most recent call first):
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CrashHandler.py", line 100 in show
  File "D:/2.6/build/inst/bin/cura_app.py", line 44 in exceptHook
  File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 659 in run
  File "D:/2.6/build/inst/bin/cura_app.py", line 70 in <module>
  File "D:\env\2.6\inst\lib\site-packages\cx_Freeze\initscripts\Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "D:\env\2.6\inst\lib\site-packages\cx_Freeze\initscripts\__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

fieldOfView, 2017-06-29:
Can you make the .obj file available for download?

AB-Normal1960, 2017-06-29:
Here is a link to the file:
https://drive.google.com/open?id=0B080dKdRrVXgTXpMeVpQYXUxWEk

nallath, 2017-06-30:
I was able to load the model, but there are a large number of mistakes with it. It could be that on different operating systems (I'm on linux) it can cause issues.

AB-Normal1960, 2017-06-30:
So, it's not printable then?

nallath, 2017-06-30:
Probably not. It will need extensive fixing.

Ghostkeeper, 2017-07-03:
> MemoryError

No, you're just running out of memory...

BagelOrb, 2017-07-21:
Bug is traced to users PC limitations

closed 2017-07-21 by BagelOrb