Ultimaker/Cura #2046: Cura is not slicing

maukcc, 2017-06-29:
Cura 2.5.99 master 
OS ubuntu 16.04

Everything is ok, the program just does not slice since yesterdays update.

[curalog.txt](https://github.com/Ultimaker/Cura/files/1111729/curalog.txt)

probably caused by:
2017-06-29 12:50:19,647 - DEBUG - UM.Backend.Backend._backendLog [90]: [Backend] [ERROR] Trying to retrieve unregistered setting with no value given: 'xy_offset_layer_0'
2017-06-29 12:50:20,267 - INFO - UM.Backend.Backend._onSocketError [196]: Backend crashed or closed.

fieldOfView, 2017-06-29:
Could you check if this change made it to your fdmprinter.def.json?
https://github.com/Ultimaker/Cura/pull/2031/commits/d1211e395135bde93ca54fc330bf89e3fa33750e#diff-be0cfced2c55852f4bce442bd971db52

maukcc, 2017-06-29:
no it was not in there
also "machine_nozzle_id" was not in there.
With both of them in, it is now working.
@thopiekar could you check what is going on?
Thanks

thopiekar, 2017-07-04:
1. No support for master
2. Updated my PPAs (stable and master) days before. Should slice again.. Had the same issue but never the time to update the PPA.