Ultimaker/Cura #2047: Cura 2.6.1 and Linux.

Fl1pp3d0ff, 2017-06-29:
After upgrading to 2.6.1 under Ubuntu from the repository, Cura now refuses to start. (PLEASE don't make me go back to Slic3r! Cura works SO much better!)

If started from the launcher, the flash screen pops up and disappears.

If started from the commandline, I get this:

`2017-06-29 11:34:06,752 - CRITICAL - cura.CrashHandler.show [39]: An uncaught exception has occurred!
2017-06-29 11:34:06,753 - CRITICAL - cura.CrashHandler.show [42]: Traceback (most recent call last):
2017-06-29 11:34:06,753 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/bin/cura", line 69, in <module>
2017-06-29 11:34:06,754 - CRITICAL - cura.CrashHandler.show [42]:     app = cura.CuraApplication.CuraApplication.getInstance()
2017-06-29 11:34:06,754 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/UM/Application.py", line 316, in getInstance
2017-06-29 11:34:06,754 - CRITICAL - cura.CrashHandler.show [42]:     Application._instance = cls()
2017-06-29 11:34:06,754 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 195, in __init__
2017-06-29 11:34:06,754 - CRITICAL - cura.CrashHandler.show [42]:     super().__init__(name = "cura", version = CuraVersion, buildtype = CuraBuildType)
2017-06-29 11:34:06,754 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 110, in __init__
2017-06-29 11:34:06,755 - CRITICAL - cura.CrashHandler.show [42]:     self._loadPlugins()
2017-06-29 11:34:06,755 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 490, in _loadPlugins
2017-06-29 11:34:06,755 - CRITICAL - cura.CrashHandler.show [42]:     self._plugin_registry.loadPlugins()
2017-06-29 11:34:06,755 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/UM/PluginRegistry.py", line 199, in loadPlugins
2017-06-29 11:34:06,755 - CRITICAL - cura.CrashHandler.show [42]:     plugin_data = self.getMetaData(plugin_id)
2017-06-29 11:34:06,755 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/UM/PluginRegistry.py", line 219, in getMetaData
2017-06-29 11:34:06,756 - CRITICAL - cura.CrashHandler.show [42]:     if not self._populateMetaData(plugin_id):
2017-06-29 11:34:06,756 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/lib/python3/dist-packages/UM/PluginRegistry.py", line 334, in _populateMetaData
2017-06-29 11:34:06,756 - CRITICAL - cura.CrashHandler.show [42]:     raise InvalidMetaDataError(plugin_id)
2017-06-29 11:34:06,756 - CRITICAL - cura.CrashHandler.show [42]: UM.PluginError.InvalidMetaDataError: Invalid metadata for plugin OrientationPlugin
`
Cura is starting from the correct folder, etc...

OS Is KUbuntu 17.04 "Zesty", just updated to the latest everything.

Any and all help would be appreciated, because I REALLY hate Slic3r...

Fl1pp3d0ff, 2017-06-29:
Removing all "extra" plugins, including the octoprint plugin, after deleting all the configs allows Cura to start.
It seems the issue is with the orientation plugin, but I'm not versed enough in python to suss what's causing it.

nallath, 2017-06-29:
Use the appimage. The PPA is not officially supported by us.

thopiekar, 2017-07-05:
If the issue still exists, please reopen a issue ticket in Cura-packaging.

closed 2017-07-05 by thopiekar