Ultimaker/Cura #2048: feature request: synchronise all extruder1 settings to extruder2

UltiArjan, 2017-06-29:
I find myself many times checking and double checking if I made adjustments to the correct extruder. I would love to have a "copy all" button to synchronize the nozzle2 settings with the nozzle1 settings.

nallath, 2017-06-30:
Would you then only want to copy the settings that you made yourself? Also copy the settings saved in a quality_changes profile?

UltiArjan, 2017-06-30:
Both would work for me, but to keep things simple a "synchronize all" would be  the most simple I guess.  I would prefer a "synchronize all". 

You could add some kind of warning when people sync 2 totally different setups, like build and support core, or 0.4 and 0.8 core, but you should be able to overrule the warning.

zerspaner-gerd, 2017-07-04:
**+1**
But possibility to do this in both directions!
Extruder 2 to 1

Ghostkeeper, 2017-07-05:
Since we support up to 16 extruders, it'd probably be more like "copy to all extruders".

DDDirk, 2017-07-05:
Or preferably let the user specify to which extruders to copy. If (in the future) that many extruders are used, there may be some exotic materials that need completely different settings. If there are 4 groups of similar materials that each occupy 4 extruders then 'copy to all' will likely be useless.
A further filter may be for specific settings, specifying which settings of an extruder to copy to which extruders.
Whether to do this now or keep it simple depends on how easy a simpler version can be 'upgraded' later.

awhiemstra, 2017-07-05:
We actually had some designs from our UX guys that would make it a lot easier to deal with these. Unfortunately, it seems they abandoned that in favour of stuff like a top bar.

Note that you can already do this on a per-setting basis, we just do not have a "global" option currently.

Ghostkeeper, 2017-07-05:
It was not abandoned, merely delayed, I think.

thesec, 2017-09-24:
+23 from #2448

Schwebewerk, 2017-11-11:
+1

i am unsure in which of these referenced threads i should best comment, but yes please. Following the User Interface Design, i see multiple extruder tabs. Each tab allows me to select a nozzle diameter and a material. The profile selection comes below all of this. Therefore it is counter intuitive, that changing a profile at this point has any influence on other extruders. This might be also related to another hidden linkage between material types and profiles [#2767](https://github.com/Ultimaker/Cura/issues/2767)

closed 2018-11-22 by Appesteijn