Ultimaker/Cura #2049: Cura 2.6.0 "Printer Not Ready" when printing over network after a print has finished

IRockButImSad, 2017-06-29:
After the last few prints done over the network when they have finished I go to start a new project, without closing Cura, and get a dialogue box that says, "Printer Not Ready". If I close out of Cura then reopen and try, it will print with no problems. Wondering if this is a known issue or not?

nallath, 2017-06-30:
Could you manually go to: {printer_ip}/api/v1/printer and post the result? 

It's an issue that i've heard about before, but it's proven to be difficult to reproduce (and if I can't reproduce it, it's also hard to fix)

IRockButImSad, 2017-06-30:
**Prior to Initial Print:**
{"bed": {"temperature": {"current": 22.4, "target": 0.0}}, "beep": {}, "diagnostics": {}, "heads": [{"acceleration": 3000.0, "extruders": [{"active_material": {"GUID": "506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9", "guid": "506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9", "length_remaining": -1.0}, "feeder": {"acceleration": 3000.0, "jerk": 5.0, "max_speed": 45.0}, "hotend": {"id": "AA 0.4", "offset": {"state": "valid", "x": 0.0, "y": 0.0, "z": 0.0}, "temperature": {"current": 25.1, "target": 0.0}}}, {"active_material": {"GUID": "fe15ed8a-33c3-4f57-a2a7-b4b78a38c3cb", "guid": "fe15ed8a-33c3-4f57-a2a7-b4b78a38c3cb", "length_remaining": -1.0}, "feeder": {"acceleration": 3000.0, "jerk": 5.0, "max_speed": 45.0}, "hotend": {"id": "BB 0.4", "offset": {"state": "valid", "x": 18.0, "y": 0.0, "z": 1.5340299999999996}, "temperature": {"current": 25.2, "target": 0.0}}}], "fan": 0.0, "jerk": {"x": 20.0, "y": 20.0, "z": 0.4}, "max_speed": {"x": 300.0, "y": 300.0, "z": 40.0}, "position": {"x": 210.69409, "y": 172.77719, "z": 16.02}}], "led": {"brightness": 100.0, "hue": 0.0, "saturation": 0.0}, "network": {"ethernet": {"connected": true, "enabled": true}, "wifi": {"connected": false, "enabled": false, "ssid": "UM-NO-HOTSPOT-NAME-SET"}, "wifi_networks": []}, "status": "idle"}
**After Initial Print before clearing the Model:**
{"bed": {"temperature": {"current": 24.7, "target": 0.0}}, "beep": {}, "diagnostics": {}, "heads": [{"acceleration": 3000.0, "extruders": [{"active_material": {"GUID": "506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9", "guid": "506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9", "length_remaining": -1.0}, "feeder": {"acceleration": 3000.0, "jerk": 5.0, "max_speed": 45.0}, "hotend": {"id": "AA 0.4", "offset": {"state": "valid", "x": 0.0, "y": 0.0, "z": 0.0}, "temperature": {"current": 28.0, "target": 0.0}}}, {"active_material": {"GUID": "fe15ed8a-33c3-4f57-a2a7-b4b78a38c3cb", "guid": "fe15ed8a-33c3-4f57-a2a7-b4b78a38c3cb", "length_remaining": -1.0}, "feeder": {"acceleration": 3000.0, "jerk": 5.0, "max_speed": 45.0}, "hotend": {"id": "BB 0.4", "offset": {"state": "valid", "x": 18.0, "y": 0.0, "z": 1.5553499999999998}, "temperature": {"current": 27.9, "target": 0.0}}}], "fan": 0.0, "jerk": {"x": 20.0, "y": 20.0, "z": 0.4}, "max_speed": {"x": 300.0, "y": 300.0, "z": 40.0}, "position": {"x": 210.69409, "y": 172.77719, "z": 6.72}}], "led": {"brightness": 100.0, "hue": 0.0, "saturation": 0.0}, "network": {"ethernet": {"connected": true, "enabled": true}, "wifi": {"connected": false, "enabled": false, "ssid": "UM-NO-HOTSPOT-NAME-SET"}, "wifi_networks": []}, "status": "idle"}
**After Clearing the Model and Inserting the new Model and selecting Print over Network:**
{"bed": {"temperature": {"current": 24.6, "target": 0.0}}, "beep": {}, "diagnostics": {}, "heads": [{"acceleration": 3000.0, "extruders": [{"active_material": {"GUID": "506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9", "guid": "506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9", "length_remaining": -1.0}, "feeder": {"acceleration": 3000.0, "jerk": 5.0, "max_speed": 45.0}, "hotend": {"id": "AA 0.4", "offset": {"state": "valid", "x": 0.0, "y": 0.0, "z": 0.0}, "temperature": {"current": 27.6, "target": 0.0}}}, {"active_material": {"GUID": "fe15ed8a-33c3-4f57-a2a7-b4b78a38c3cb", "guid": "fe15ed8a-33c3-4f57-a2a7-b4b78a38c3cb", "length_remaining": -1.0}, "feeder": {"acceleration": 3000.0, "jerk": 5.0, "max_speed": 45.0}, "hotend": {"id": "BB 0.4", "offset": {"state": "valid", "x": 18.0, "y": 0.0, "z": 1.5553499999999998}, "temperature": {"current": 27.5, "target": 0.0}}}], "fan": 0.0, "jerk": {"x": 20.0, "y": 20.0, "z": 0.4}, "max_speed": {"x": 300.0, "y": 300.0, "z": 40.0}, "position": {"x": 210.69409, "y": 172.77719, "z": 6.72}}], "led": {"brightness": 100.0, "hue": 0.0, "saturation": 0.0}, "network": {"ethernet": {"connected": true, "enabled": true}, "wifi": {"connected": false, "enabled": false, "ssid": "UM-NO-HOTSPOT-NAME-SET"}, "wifi_networks": []}, "status": "idle"}
**After Closing Cura and Setting up new Print:**
{"bed": {"temperature": {"current": 24.0, "target": 0.0}}, "beep": {}, "diagnostics": {}, "heads": [{"acceleration": 3000.0, "extruders": [{"active_material": {"GUID": "506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9", "guid": "506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9", "length_remaining": -1.0}, "feeder": {"acceleration": 3000.0, "jerk": 5.0, "max_speed": 45.0}, "hotend": {"id": "AA 0.4", "offset": {"state": "valid", "x": 0.0, "y": 0.0, "z": 0.0}, "temperature": {"current": 27.5, "target": 0.0}}}, {"active_material": {"GUID": "fe15ed8a-33c3-4f57-a2a7-b4b78a38c3cb", "guid": "fe15ed8a-33c3-4f57-a2a7-b4b78a38c3cb", "length_remaining": -1.0}, "feeder": {"acceleration": 3000.0, "jerk": 5.0, "max_speed": 45.0}, "hotend": {"id": "BB 0.4", "offset": {"state": "valid", "x": 18.0, "y": 0.0, "z": 1.5553499999999998}, "temperature": {"current": 27.3, "target": 0.0}}}], "fan": 0.0, "jerk": {"x": 20.0, "y": 20.0, "z": 0.4}, "max_speed": {"x": 300.0, "y": 300.0, "z": 40.0}, "position": {"x": 210.69409, "y": 172.77719, "z": 6.72}}], "led": {"brightness": 100.0, "hue": 0.0, "saturation": 0.0}, "network": {"ethernet": {"connected": true, "enabled": true}, "wifi": {"connected": false, "enabled": false, "ssid": "UM-NO-HOTSPOT-NAME-SET"}, "wifi_networks": []}, "status": "idle"}

Ghostkeeper, 2017-07-05:
They are all four exactly the same, apart from the temperature and the Z position in the first one.

Normally it should not indicate a status of `"idle"` if the build plate is not clear.

Ghostkeeper, 2017-07-06:
@MarcoTvM Could you investigate? It seems like the status returned by the printer API might not be updating.

ckielstra, 2017-07-13:
What is the firmware version number? I tried with Cura 2.6.1 and UM3 firmware v3.7 and couldn't reproduce the issue. The printer status is updating as expected.

Note that the printer state should report "printing" when the print is finished but still awaits the user to clear the build plate. Fact that the above log shows the "idle" status could indicate a bug which already has been solved in the v3.7 firmware.

mickyblueeyes, 2017-07-27:
HI  I have a new Ultimaker 3 which I installed last weekend it is up to the latest stable firmware and i'm using Cura 2.6.2 and I have the same problem.

ckielstra, 2017-08-09:
@mickyblueeyes you didn't mention the used firmware version but it must be a 3.6 version, the v3.7 was released one day after your post.
The posted log shows the printer in 'idle' while waiting for the buildplate to be cleared, that's wrong. I again tested in v3.7 and here it always shows correctly as 'printing'. One change we made was replacing the webserver in the UM3 and perhaps the former v3.6 was caching old data?

Since I cant reproduce the issue I'll close it. When someone finds it happening again, then post here and we'll look into it again.

closed 2017-08-09 by ckielstra