Ultimaker/Cura #2050: 2.6.0 & 2.6.1 Multiple objects not all printing on the first layer

dfaszer, 2017-06-30:
I have a build plate of 6 items. Three of the items are identical half spheres with a hole on the flat, downward side, while the other three are more unique. I am printing with a skirt and I have tried an initial layer height of .2mm and .1mm. I found this issue in 2.6.0 and upgraded to 2.6.1, loaded the .mf file, and the issue was still there. I haven't tried 2.6.1 starting from scratch tho.

I noticed the issue when I started the print. The three unique items printed just fine on the first layer with the skirt going around the perimeter as if there were only three items. The half spheres didn't start printing until layer 2 and one of the spheres that were close the the unique parts actually printed over top the skirt.

I initially had the per model settings selected to have support on one of the items and not the rest. I removed that setting and it is still happening. I then tried support on for all of the components. On layer one, there is support for the hole for the half spheres, but no model layer.

I uploaded the .3mf file and some pictures if they will help. I wan't able to upload in the normal format, so I had to add ".txt" to the end of the file name. I also tried uploading the gcode, but it was too big. The first two images are without support, while the second two are with support.

[PI_Cold Brew.3mf.txt](https://github.com/Ultimaker/Cura/files/1114042/PI_Cold.Brew.3mf.txt)
[image]
[image]
[image]
[image]

nallath, 2017-06-30:
That's weird. I loaded your model, but it prints the half spheres on layer one for me. Do you have the "drop objects to buildplate" enabled?

dfaszer, 2017-06-30:
Yes, I have the drop to build plate enabled (actually don't know how to turn it off). When I raise up a file, it drops right back to the build plate. I loaded up the file that I attached onto a different computer, and it has the same issue.

When I first put the models on, I used the "per model setting" so that I can do a higher density on one of the "Y"'s and to also add support to one of the "Y"'s. Could that have caused an issue?

I'll try importing the files fresh from a new session of Cura and report back.

nallath, 2017-06-30:
The per setting models should not have an influence on this.

dfaszer, 2017-06-30:
So I'm on a different computer today than I was last night. I updated Cura to 2.6.1 and manually loaded each model starting with a fresh session of Cura. I didn't change anything settings wise, but I believe it should have used the last saved settings. When I checked the layers, it is still having the same problem.

I tried changing to one of the default profiles (Coarse Quality 0.4mm). There was a pop-up box asking if I wanted to keep any of the settings. I told it to reset all in hopes of starting from scratch. Still have the same layer problem. I switched back to my main profile, and now it seems to be working as normal.

dfaszer, 2017-06-30:
I think I found the error.

If I put the initial layer height at .3mm, it works just fine. If I put the initial layer height at .2mm or .1mm, then it has the error.

I've printed with multiple objects like this before with an initial layer height of .1mm on a regular basis and never run across this issue before.

dfaszer, 2017-06-30:
Here is the link for the models if for some reason it might be a model specific issue.
https://www.instructables.com/id/Coffee-Cold-Brewer/

dfaszer, 2017-06-30:
I am seeing the error with the default profiles as well, but with those, it errors out at different initial layer heights. For the coarse quality 0.4mm, it errors at an initial layer height of .3mm, but works at .4mm. For the fine quality 0.1mm, it errors at an initial height of .1mm, but works at .2mm.

dfaszer, 2017-06-30:
So doing some more testing, the issue also seems to be specific to the half spheres. These aren't my models, but considering the others work just fine, I would imagine that the spheres would be made the same way (no weird gap under the model).

retegic, 2017-07-03:
Experiencing the same issues here as well with multiple objects. First object prints fine, second one has some funny first layer issues.

Ghostkeeper, 2017-07-05:
I suspect that this is the issue again where the frond-end doesn't properly drop the model entirely to the build plate. Normally the engine removes layers that are empty afterwards anyway so we don't typically find this in our tests if we test with only one model.

Ghostkeeper, 2017-07-05:
Doesn't reproduce on my computer, by the way, even with 0.1mm layer height. Maybe something else is different.

BagelOrb, 2017-07-14:
This is most probably cause by small artifacts in the model which cause the first layer to be empty or too small to print.

This should be resolved by making an engine setting in the frontend to drop models on the buildplate before support is added.

Ghostkeeper, 2017-12-15:
This is fixed for Cura 3.2.

closed 2017-12-15 by Ghostkeeper

logstonc, 2018-03-07:
I had a similar problem using Cura 2.7, 3.0 and 3.2 with support enabled, all had the same problem with drop to build plate enabled or not... I had to move all parts that were not printing on the first layer down manually a single layer height, and success https://youtu.be/mfvotlvvj_o