Ultimaker/Cura #2052: Group/Ungroup

Guisao5, 2017-06-30:
Group and Ungroup not active in menu

nallath, 2017-06-30:
I just tried it; When I have multiple objects selected, the group & merge options are available. When i have a group selected, the ungroup is active.

Guisao5, 2017-06-30:
Correct, but when the stl comes grouped do not ungroup, Cure 15 ungroups.

nallath, 2017-06-30:
Ah, you mean split into parts? That's indeed not implemented for the new Cura yet.

PJParish, 2017-07-23:
I'll follow it here.  Thanks.

Ghostkeeper, 2018-02-05:
Our project manager just removed this feature from our planning because it was not modified for 12 weeks. Or rather, it would be next week.

ScottWell1, 2018-02-05:
The feature is still desired.  It's a big loss for a lot of people coming from legacy Cura, particularly those with small build plates who need to split plated models into parts in order to print them.

cscott, 2018-02-05:
Especially folks using services like thingiverse to get their STLs, who
don't necessarily have the ability to "not plate" unrelated parts.​  They
are at the mercy of whoever uploaded the STL (and that person's build plate
size).

fieldOfView, 2018-08-24:
I am working on it as a plugin for the next version of Cura:
[image]

fieldOfView, 2018-10-06:
"Split into parts" is now available as part of "Mesh Tools" in the Toolbox of Cura 3.5

closed 2018-10-08 by nallath