Ultimaker/Cura #2053: Layer preview not working for dual extrusion printers

unknownlamer, 2017-06-30:
Layer previews fail while processing layers for dual extrusion printers for some weeks now with the following errors:

2017-06-30 10:44:44,417 - ERROR - UM.Logger.logException [76]: Exception: Job <CuraEngineBackend.ProcessSlicedLayersJob.ProcessSlicedLayersJob object at 0x7fdaa4450a20> caused an exception
2017-06-30 10:44:44,419 - ERROR - UM.Logger.logException [80]: Traceback (most recent call last):
2017-06-30 10:44:44,419 - ERROR - UM.Logger.logException [80]:   File "/home/clinton/src/vcs/cura2-dev_26/uranium/UM/JobQueue.py", line 125, in run
2017-06-30 10:44:44,420 - ERROR - UM.Logger.logException [80]:     job.run()
2017-06-30 10:44:44,420 - ERROR - UM.Logger.logException [80]:   File "/home/clinton/src/version-controlled/cura2-dev_26/user-interface/cura/../plugins/CuraEngineBackend/ProcessSlicedLayersJob.py", line 178, in run
2017-06-30 10:44:44,420 - ERROR - UM.Logger.logException [80]:     color_code = material.getMetaDataEntry("color_code", default="#e0e000")
2017-06-30 10:44:44,420 - ERROR - UM.Logger.logException [80]: AttributeError: 'NoneType' object has no attribute 'getMetaDataEntry'

Loading the exported gcode file renders fine. Single extrusion prints using the same material and quality (but different printer definition) do not have any problem with the layer preview.

Using Cura master e98a6d76ba31a260e79fde2bc5abfe0e647f3bf2, Uranium 80c6817fed4087d38ba3f2918a8fdeb4ce100eea, Arcus e126352f14c10d55a4d094554d15d29438bceb75, python3-pyqt5 5.8.2

Ghostkeeper, 2017-07-06:
I don't have this issue, it doesn't reproduce. Looking at the code where this exception occurs I see a bit of an old technique to find the material, which can now be easier and more robust. So I've updated it. For me it still works the same but it might not crash any more.

Hopefully this fixes it for you: https://github.com/Ultimaker/Cura/commit/bb68f488dbf9af533b59b11ae3908e5c9b137e3a

Could you pull Cura and Uranium, and try again?

unknownlamer, 2017-07-06:
Hey,

Fresh pull works fine with dual extrusion printers now (although it also reset my printer and material definitions again so re-creating those might have something to do with it). Thanks for the quick fix.

closed 2017-07-06 by unknownlamer