Ultimaker/Cura #2058: Add Flow to Materials

ethanspitz, 2017-07-03:
I feel like Flow should be part of materials. Considering part my my calibration workflow is to recalibrate the flow per material, I'd like to be able to set that per material. Thoughts?

Ghostkeeper, 2017-07-06:
We are sort of limited by the XML material definition, which is not completely under Cura's control.

We had plans to add a cura namespace to the XML files so that Cura could add settings at will. But I see that the issue we had about that in our tracker was deleted for being too old, so I just had to re-create that...

fieldOfView, 2017-07-06:
> We are sort of limited by the XML material definition, which is not completely under Cura's control.

Which is kind of ridiculous, because it is an Ultimaker defined standard that noone else uses.

nallath, 2017-07-06:
I don't think it's that hard to add the flow to the material profile. Granted, we'd have to discuss it with some developers outside the Cura team, but that's not a big issue (I'd even go as far as to say it's always a good idea to discuss changes to a spec with multiple people)

Liger0, 2018-11-21:
Is there any updates? The flow doesn't have any reason to stay part of a profile instead of a profile.

fieldOfView, 2020-03-06:
For the benefit of those stumbling upon this issue: Check out the Material Settings plugin from the Marketplace. It will let you most settings to the Material pane of the preferences, saved per-material. Do note however that if you change the same setting value in the sidebar, that value will override what is set in the Material

Ghostkeeper, 2020-03-06:
For the record, this is still something I'm keeping in the back of the mind since it's quite a common request. But it's hard to make material profiles properly understandable to the user and fieldOfView's plug-in does as good a job as any.

nallath, 2020-11-03:
I think this issue can be closed; There is a way to do this via the material settings plugin.

closed 2020-11-03 by nallath