Ultimaker/Cura #2060: Cura master branch crash

smartavionics, 2017-07-03:
03/07/2016 - 1530

Currently trying the master branches of Uranium, Cura & CuraEngine and get this:

2017-07-03 15:32:27,601 - DEBUG - UM.Settings.ContainerRegistry.load [213]: Loading data into container registry took 11.23727798461914 seconds
2017-07-03 15:32:27,603 - DEBUG - UM.Backend.Backend._logSocketState [172]: Socket state changed to Connected
2017-07-03 15:32:27,603 - DEBUG - UM.Backend.Backend._onSocketStateChanged [162]: Backend connected on port 49674
2017-07-03 15:32:27,604 - DEBUG - UM.Controller.setActiveView [84]: Setting active view to SolidView
2017-07-03 15:32:27,760 - CRITICAL - cura.CrashHandler.show [39]: An uncaught exception has occurred!
2017-07-03 15:32:27,762 - CRITICAL - cura.CrashHandler.show [42]: Traceback (most recent call last):
2017-07-03 15:32:27,762 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/local/bin/cura", line 73, in <module>
2017-07-03 15:32:27,762 - CRITICAL - cura.CrashHandler.show [42]:     app.run()
2017-07-03 15:32:27,762 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/local/lib/python3/dist-packages/cura/CuraApplication.py", line 620, in run
2017-07-03 15:32:27,762 - CRITICAL - cura.CrashHandler.show [42]:     self._volume = BuildVolume.BuildVolume(root)
2017-07-03 15:32:27,762 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/local/lib/python3/dist-packages/cura/BuildVolume.py", line 102, in __init__
2017-07-03 15:32:27,763 - CRITICAL - cura.CrashHandler.show [42]:     Application.getInstance().getMachineManager().activeQualityChanged.connect(self._onStackChanged)
2017-07-03 15:32:27,763 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/local/lib/python3/dist-packages/cura/CuraApplication.py", line 670, in getMachineManager
2017-07-03 15:32:27,763 - CRITICAL - cura.CrashHandler.show [42]:     self._machine_manager = MachineManager.createMachineManager()
2017-07-03 15:32:27,763 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/local/lib/python3/dist-packages/cura/Settings/MachineManager.py", line 1123, in createMachineManager
2017-07-03 15:32:27,763 - CRITICAL - cura.CrashHandler.show [42]:     return MachineManager()
2017-07-03 15:32:27,763 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/local/lib/python3/dist-packages/cura/Settings/MachineManager.py", line 97, in __init__
2017-07-03 15:32:27,763 - CRITICAL - cura.CrashHandler.show [42]:     self.setActiveMachine(active_machine_id)
2017-07-03 15:32:27,764 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/local/lib/python3/dist-packages/cura/Settings/MachineManager.py", line 347, in setActiveMachine
2017-07-03 15:32:27,764 - CRITICAL - cura.CrashHandler.show [42]:     Application.getInstance().setGlobalContainerStack(containers[0])
2017-07-03 15:32:27,764 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/local/lib/python3/dist-packages/UM/Application.py", line 152, in setGlobalContainerStack
2017-07-03 15:32:27,764 - CRITICAL - cura.CrashHandler.show [42]:     self.globalContainerStackChanged.emit()
2017-07-03 15:32:27,764 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/local/lib/python3/dist-packages/UM/Signal.py", line 212, in emit
2017-07-03 15:32:27,764 - CRITICAL - cura.CrashHandler.show [42]:     self.__performEmit(*args, **kwargs)
2017-07-03 15:32:27,764 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/local/lib/python3/dist-packages/UM/Signal.py", line 315, in __performEmit
2017-07-03 15:32:27,765 - CRITICAL - cura.CrashHandler.show [42]:     func(dest, *args, **kwargs)
2017-07-03 15:32:27,765 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/local/lib/python3/dist-packages/cura/BuildVolume.py", line 490, in _onStackChanged
2017-07-03 15:32:27,765 - CRITICAL - cura.CrashHandler.show [42]:     self._updateDisallowedAreas()
2017-07-03 15:32:27,765 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/local/lib/python3/dist-packages/cura/BuildVolume.py", line 564, in _updateDisallowedAreas
2017-07-03 15:32:27,765 - CRITICAL - cura.CrashHandler.show [42]:     result_areas = self._computeDisallowedAreasStatic(disallowed_border_size, used_extruders) #Normal machine disallowed areas can always be added.
2017-07-03 15:32:27,765 - CRITICAL - cura.CrashHandler.show [42]:   File "/usr/local/lib/python3/dist-packages/cura/BuildVolume.py", line 730, in _computeDisallowedAreasStatic
2017-07-03 15:32:27,765 - CRITICAL - cura.CrashHandler.show [42]:     offset_y = -extruder.getProperty("machine_nozzle_offset_y", "value")
2017-07-03 15:32:27,766 - CRITICAL - cura.CrashHandler.show [42]: TypeError: bad operand type for unary -: 'NoneType'

fieldOfView, 2017-07-03:
https://github.com/Ultimaker/Cura/commit/319559740dd8c8c2f9ffad637bee43cef9345073 needs to be reverted

smartavionics, 2017-07-03:
Ah, yes, I can see what's gone wrong there... thanks.

nallath, 2017-07-03:
Done.

closed 2017-07-03 by nallath