Ultimaker/Cura #2061: Everything prints as a support on Linux (both 2.6.1 and 2.4.0)--MacOS works fine

ppesek, 2017-07-03:
I am having an odd issue where if I turn on supports and slice in Cura under Linux, the whole model prints as flimsy support-like material.  I have narrowed down the problem to only linux (2.6.1 and 2.4.0 AppImages under Ubuntu 16.04).  

Using MacOS Cura 2.4.0, I was not able to reproduce the problem using the same STL file.  Attached are example files that cause the issue to occur and a picture of the result.

[CuraIssue.zip](https://github.com/Ultimaker/Cura/files/1120269/CuraIssue.zip)

ppesek, 2017-07-03:
I should probably add I printed using a Prusa i3 Mk2 with the latest firmware (3.11).

smartavionics, 2017-07-03:
Hi, I am using the master branch on linux and it appears to slice OK, here's a typical cross section...
[image]

ppesek, 2017-07-03:
It seems like you changed something in the settings to get that screenshot as the supports are closer together than I had them (not that it made a difference--started with something like what you had)--did you load the brokensupports profile I included?

smartavionics, 2017-07-03:
Oops, no, I used a different profile. Here's another sample using your profile...
[image]

ppesek, 2017-07-03:
Yeah...the display is different on my end--notice the texture difference?  Not sure what is causing that as it prints the same way...

[image]

smartavionics, 2017-07-03:
You are displaying the travels (blue) but I did not? It's quite hard to see the detail on your screenshot.

ppesek, 2017-07-03:
Notice the texture on the red part though?  That remains if I turn blue off and seems to represent the main difference.  It actually looks like that if I turn supports off (though they are necessary to print this)--so maybe the issue is unrelated to supports.  My initial theory was that it was printing the whole thing as 'like a support structure'--but maybe not...

smartavionics, 2017-07-03:
To me, that texture just looks like aliasing in the graphics? Does it alter as you rotate the view?

ppesek, 2017-07-03:
It does--guess you are right...  Still not sure why my models are printing as flexible mush suddenly.  The same STL sliced on a Mac printed fine (and with Slic3r on Linux also).

Ghostkeeper, 2017-07-07:
The specialised profile included in the .zip doesn't seem to be used in your MacOS g-code. It specifies a layer height of 0.2mm. The object is about 4cm high. In the MacOS g-code there are 411 layers and in the Linux g-code there are 206 layers, so the MacOS clearly uses 0.1mm layer height and the Linux g-code uses 0.2mm layer height. That's the difference you're seeing.

So it looks like the 0.2mm profile just isn't working properly for your printer. Have you tried reducing the speed a bit? Perhaps the nozzle is a bit half-clogged and isn't keeping up with the double extrusion rate that's required for the greater layer height?

BagelOrb, 2017-07-14:
This seems to be a hardware problem unrelated to Cura.

closed 2017-07-14 by BagelOrb