Ultimaker/Cura #2062: Cura 2.6.1 - Unable to replace 'travel_speed' placeholder in start/end gcode in cura.log

jaseiffert, 2017-07-04:
2017-07-03 20:11:53,578 - WARNING - CuraEngineBackend.StartSliceJob.get_value [38]: Unable to replace 'fill_density' placeholder in start/end gcode
2017-07-03 20:11:53,579 - WARNING - CuraEngineBackend.StartSliceJob.get_value [38]: Unable to replace 'print_time' placeholder in start/end gcode
2017-07-03 20:11:53,579 - WARNING - CuraEngineBackend.StartSliceJob.get_value [38]: Unable to replace 'filament_amount' placeholder in start/end gcode
2017-07-03 20:11:53,580 - WARNING - CuraEngineBackend.StartSliceJob.get_value [38]: Unable to replace 'filament_weight' placeholder in start/end gcode
2017-07-03 20:11:53,581 - WARNING - CuraEngineBackend.StartSliceJob.get_value [38]: Unable to replace 'filament_cost' placeholder in start/end gcode
2017-07-03 20:11:53,582 - WARNING - CuraEngineBackend.StartSliceJob.get_value [38]: Unable to replace 'travel_speed' placeholder in start/end gcode
2017-07-03 20:11:53,582 - WARNING - CuraEngineBackend.StartSliceJob.get_value [38]: Unable to replace 'travel_speed' placeholder in start/end gcode
2017-07-03 20:11:53,586 - WARNING - CuraEngineBackend.StartSliceJob.get_value [38]: Unable to replace 'travel_speed' placeholder in start/end gcode

I noticed when I looked at the gcode file Cura produced that the travel_speed wasn't being replaced by a number so looked at the Cura.log and found these.

Cura 2.6.1 installed on Windows 10 Pro 64bit.

Doesn't work in 2.5 nor 2.3 and couldn't find where to put start/end gcode in 2.1.2 so don't know.

I'll just hardcode the gcode right now but wanted to let you know.

Thanks,

Jeffery

fieldOfView, 2017-07-04:
Use {infill_sparse_density} instead of {fill_density} and {speed_travel} instead of {travel_speed}

I don't know why {print_time} and {filament_*} don't work. They should be replaced here:
https://github.com/Ultimaker/Cura/blob/2.6/plugins/CuraEngineBackend/CuraEngineBackend.py#L464

jaseiffert, 2017-07-05:
Thanks a lot for the information. I went through it today again, cleared the log, and the only thing it really doesn't like is the travel_speed so not sure why it was giving me all those warnings. Except for travel_speed I didn't have any of the others in my start gcode.

I changed my start/end gcode to use speed_travel and it worked just fine.

Thanks for all the help,

Jeffery

fieldOfView, 2017-07-05:
Feel free to close the issue

closed 2017-07-05 by jaseiffert