Ultimaker/Cura #2066: skirt setting isnt working

40kdominator, 2017-07-04:
when I go to print build plate adhesion it resets the parameters to brim I go and change it to skirt but I go back and it turned off build plate adhesion but when I turn it back on it resets again so there is no way to use the skirt it just started yesterday

nallath, 2017-07-04:
What version did you use? 
What machine did you have active?
Could you post logs?

40kdominator, 2017-07-04:
i tried 2.6.1 and 2.3.1 i was using my own costom 3d printer i have no idea what post logs is?

Ghostkeeper, 2017-07-07:
Instructions for sharing the logs are here: https://github.com/Ultimaker/Cura#logging-issues

Perhaps you're using a dual extrusion printer and the second extruder still specifies that it wants to print with a brim.

closed 2018-02-26 by ianpaschal