Ultimaker/Cura #2067: Cura 2.6.1 Ultimaker 2 gram-calculation

Manusu90, 2017-07-04:
Hello together
Since i have updatet my cura to 2.6.1 the calculation for the grams doesn't work anymore. 
I have the Ultimaker 2 and the software version that i had before was something arround 15.4...
Is it possible to fix this bug? 

thx for the help in advance
Manusu90

nallath, 2017-07-04:
Cura doesn't know what material you are slicing for if you are using the UM2. As a result, it doesn't know the density, which is needed to calculate the weight from the volume of material extruded.
Cura 15.04 used the (incorrect) value of PLA by default.

closed 2017-07-04 by nallath

fieldOfView, 2017-07-04:
Workaround: If you go into Machine Settings and change your gcode flavor to 'Marlin (Reprap/Sprinter)' you get full control over materials in Cura, including finetuning temperatures and weight and cost calculation.

Manusu90, 2017-07-04:
Cura can calculat the length of the using material but not the weight? 
I us standard 2.85mm filament so why this is not possibel to calculat?

nallath, 2017-07-05:
Because the machine extrudes based on volume. We know how much material we need to extrude. This is the same for every material. But if you have 50 mm3 of material A, the weight is not the same as material B. 
As we don't know the material, we don't know the weight.

Manusu90, 2017-07-05:
Aaaa okey thanks for your statement. I can understand now why this is so. Thanks again