Ultimaker/Cura #2069: Feature Request: New honeycomb infill pattern

zerspaner-gerd, 2017-07-04:
# Suggestion for new, further infill pattern!
###  **The honeycomb pattern**
[image]

- **It looks good**
[image]

- There is a crucial difference to the other available infill variants, hardly overlapping lines

- Can be printed more solidly, more comfortably, more consistently, since there is no departure from the material flow

- Slightly more stability

BagelOrb, 2017-07-05:
So you mean the Full Honeycomb from S3D.
The fast honeycomb does have self-intersections..

I don't think it gives more stability, though. I think the Honeycomb is more squishy in the XY directions than triangles.

Actually I think you're the first person to give a holding argument for Honeycomb infill: the lack of overlapping lines!

DDDirk, 2017-07-05:
I have been playing with something similar, reasoning from the principle that circles (or spheres) are the strongest stuctures.
Here's a quick sketch (I really should learn to use Inkscape :) ):

[image]

A next row can then be added thus:

[image]

Etc.

This is one continuous road, so no travel moves are needed. And there are no angles, which are often a problem in prints. But it may also mean the print speed can go up.
And there are more things one can do with this, such as making the lower 'circles' smaller, which will create a curving shape. Angles are a bit trickier, but can be done by making the circles increasingly smaller, then changing the direction and making the circles bigger again. 
And the sizes of the circles can vary, to make those near the outer walls smaller, to make the structure stronger there. There are loads of possible variations on this.

I suspect this will be strong (per print time) but not very stiff. And that may often be very desirable.

But on topic, of course this can be adapted to make a honeycomb pattern by making the 'circles' hexagonal. A major drawback is that whilst most sides are then double-walled, some sides will be single-walled, namely the middle ones in the first drawing.

nickthetait, 2017-07-05:
Can you think of a specific use case where this pattern is desired and no other patterns would suffice?

DDDirk, 2017-07-06:
Well, like I said, if you want to print fast and/or you want the print to be strong but not stiff (what BagelOrb calls squishy :) ). At least, that is what I expect of this. That can be useful if you want the print to give before it breaks (eg with a brittle material). Or for protection. The mobile cases in the OP are a good example of when you would want that.
Also, the sizes of the circles can vary, making bigger ones in the middle and smaller ones near the edges. Or other arrangements for different combinations of flexibility. Or just to reduce the print time and material.

It is something I want to experiment with when I have the time (which is never :) ), but I just posted it because BagelOrb mentioned the lack of overlapping lines, by which I suppose he meant fewer crossing lines. And this has none (in principle).

zerspaner-gerd, 2017-07-06:
One short before my english is not good
> So you mean the Full Honeycomb from S3D.

Yes
> The fast honeycomb does have self-intersections..

I do not see any, here is a fast video
[Video on my DropBox](https://www.dropbox.com/s/0sgihq8vht45zi4/Infill%20Muster.mp4?dl=0)
The honeycombs are offset to each layer.

> 
> Ich glaube nicht, dass es mehr Stabilität gibt. Ich glaube, die Wabe ist in den XY-Richtungen mehr zierlich als Dreiecke.
> 
> Eigentlich denke ich, du bist die erste Person, die ein Holding-Argument für Honeycomb-Füllung gibt: den Mangel an überlappenden Linien!

I print many mechanical parts (hobbies) with Colorfabb XT and with normal Infillarten looks so
[image]

The first lines everything looks good, in the subsequent travel at all

@DDDirk Yes that is good, is based on the same as the honeycombs. For stiffness with circles, I can not say anything

> Kannst du an einen speziellen Anwendungsfall denken, wo dieses Muster gewünscht wird und keine anderen Muster ausreichen würden?

Since I have found the honeycomb pattern I print all parts with these (until now but only with S3D ;-)) No matter what material there is no problems, Infill is always fully intact
Same speed, temp as the picture above
[image]

Overlapping line are OK (believe also necessary), crossings are the ones that give negative results to me

BagelOrb, 2017-07-07:
Oh I totally misunderstood how the Fast honeycomb works. Thanks for the video!

The second layer of the Fast honeycomb is not that clear in your video.
Is it actually not filling op the full pattern on that layer?

BagelOrb, 2017-07-07:
@DDDirk please post your ideas about a different infill pattern in a new issue. This issue is about honeycomb infill.

BagelOrb, 2017-07-07:
These are the reasons I see now for using honeycomb:
* It doesn't have any overlapping lines
* It absorbs more force; it is less brittle than linear infill patterns.

zerspaner-gerd, 2017-07-09:
> The second layer of the Fast honeycomb is not that clear in your video.
> Is it actually not filling op the full pattern on that layer?

Here once again for the better visibility individual layer view
**Fast honeycomb**
[image]

**Full honeycomb**
[image]

mrjohnc, 2018-01-20:
+1 for supporting Honeycomb infill, faster infill is always welcome :)

BagelOrb, 2018-01-20:
Honeycomb infill is slower..

gnbl, 2018-05-28:
A honeycomb infill has previously been discussed in https://github.com/Ultimaker/Cura/issues/1287.
I'd also like to have this option.

LazaroFilm, 2018-06-23:
I would really love to have the option of Honeycomb infill with Cura, it's one of the only aspects that cura is really missing compared to other programs. At least allow us to import infill files and let the community program them.

getget1980, 2018-06-27:
Would really be happy to have (fast) honeycomb too :)

Sjoerrdd, 2019-12-22:
Any progress in adding Honeycomb?

smartavionics, 2019-12-22:
AFAIK, nobody is working on honeycomb infill but there should be some new infill patterns coming along. Someone has submitted a PR for truncated octahedron infill (https://github.com/Ultimaker/Cura/pull/6800) and I have recently implemented Schwarz P and Schwarz D infill patterns (see https://community.ultimaker.com/topic/30456-new-infill-patterns-in-experimental-cura-builds/) but that is currently only available in my Cura builds but I expect to submit a PR before too long.

skrutt, 2020-03-30:
YES, been waiting for this for AGES!
One more argument for hexagonal: The lack of straight lines mitigates warping. The only other option right now is gyroid.. but really, thats fun and all but seems way more silly than hexagonal! 
Not sure how strong gyroid is but probably not as good as hex.

frankk74, 2020-09-30:
Yes another feature request with an additional (valid?) reason to reconsider?

Some people are cheating :-) and not using modeling software to create "cool" vents :-) besides the "jones" do it now ....

https://p3dt.net/assets/prj-nanopim4-server/02-installation/01-the-fan.jpg

Maybe not good enough reasons but another ask? Pleaseeee :-)

LazaroFilm, 2020-10-15:
Well, this was the straw that broke the camel's back. I have now officially stopped using Cura and moved on to SuperSlicer. My initial motivation for the switch was this Honeycomb infill, then I stayed with SuperSlicer for all the awesome features it offers. Bye!

Tasty213, 2020-10-21:
Hi, if your looking for more reasons to add honeycomb infill this IOP article clearly shows that it is a very good infill to use https://iopscience.iop.org/article/10.1088/1757-899X/227/1/012033/meta

mrjohnc, 2020-10-22:
Thanks very much @Tasty213 the paper clearly states "honeycomb infill pattern can be considered the best pattern to maximise the tensile strength of 3D prints."

nallath, 2020-10-22:
> Thanks very much @Tasty213 the paper clearly states "honeycomb infill pattern can be considered the best pattern to maximise the tensile strength of 3D prints."

Although slightly before that: " itstorsional  stressparameters  are  probably  much  worse.  ". It's based on that assumption that next statement is made. 

The paper also claims that it used CuraEngine for slicing, but it doesn't even have honeycomb infill. 

Last but not least; They tested a single dogbone per infill pattern. There are still quite a bit of variations between prints, so they should have made multiple of these tensile bars and averaged the results.

ontheronix, 2020-10-22:
The paper has it's limitations indeed. I would think 3D-honeycomb-like patterns will make for high torsional strengths. (We are looking for strong infill patterns for windturbineblades - www.fromwastetowind.com)

fieldOfView, 2020-10-22:
> The paper also claims that it used CuraEngine for slicing, but it doesn't even have honeycomb infill.

A long, long time ago, Type A machines made a modified Cura 15.4.x (Type A Cura 1.x) which had honeycomb support. It seems to be still available for download here:
https://bitbucket.org/typeamachines/cura-type-a/downloads/
Note that this is a 4 year old version of Cura.

BagelOrb, 2020-11-04:
One good argument for implementing hex honeycomb infill:
https://youtu.be/thOifuHs6eY

Tasty213, 2020-12-01:
> Although slightly before that: " itstorsional stressparameters are probably much worse. ". It's based on that assumption that next statement is made.
> Last but not least; They tested a single dogbone per infill pattern. There are still quite a bit of variations between prints, so they should have made multiple of these tensile bars and averaged the results.
I admit that the paper is slightly flawed but how do you expect more research to be done if there isn't a widely available slicer that supports it? If its bad then people will just figure not to use it and it can be removed.

smartavionics, 2020-12-01:
For a version of Cura that provides a honeycomb infill/support pattern see https://github.com/smartavionics/Cura/releases/tag/20201121.

daniel-finger, 2022-07-10:
Actually I like print some Objects without Top and Bottom Layers and would like to use a Honeycomb Structure for an aesthetic look

smartavionics, 2022-07-10:
Since my previous comment, I have also implemented Hilbert curve infill/support pattern and it can also be used for skin too. Not that useful except for aesthetics. See https://github.com/smartavionics/Cura/releases if you are interested.

[image]