Ultimaker/Cura #2070: Question about dual color and merging

jrdiesel76, 2017-07-05:
Windows 10, Cura 2.6.1
Last night I printed my first dual color print, successfully.  Tonight I try to print a different 2 color .stl, I load both parts of the model, select part 1 to extruder 1 (blue), part 2 for extruder 2 (black), select all, merge...now the entire model is printing either all from extruder 1 OR extruder 2.  Once they merge, it seems to forget about the 2 extruders.  I feel like I'm missing a step, but can't quite put my finger on it...

The model is nefertiti-dual-color-by-frederic_poignie on thingiverse, https://www.thingiverse.com/thing:1377950

Thanks in advance.

closed 2017-07-05 by jrdiesel76

jrdiesel76, 2017-07-05:
Figured it out. :)

fieldOfView, 2017-07-05:
For the benefit of people reaching this page through a search engine, care to explain how you solved it?

jrdiesel76, 2017-07-06:
What I was doing wrong was, after I loaded model 1 and picked extruder 1 on the left side of the screen, and loaded model 2 and picked extruder 2 on the left side of the screen, I right clicked and merged the models, that's where i should have stopped.  BUT after merging I hit extruder 1 on the left of the screen which switched the entire merged model to extruder 1, cancelling out my previous extruder choices.  If I wanted to check any settings, I should have checked using the menus on the right side of the screen for extruder 1 and 2.