Ultimaker/Cura #2071: Working repository for Ubuntu 16.04 and later?

isbm, 2017-07-05:
Currently installing Cura from @thopiekar 's repos — https://launchpad.net/~thopiekar/+archive/ubuntu/cura — it is entirely broken on 16.04:
```
$ apt install cura
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies.
 cura : Depends: python3-uranium but it is not going to be installed
        Recommends: python3-numpy-stl but it is not installable
E: Unable to correct problems, you have held broken packages.
```

The Uranium build system package (why is even needed there??) is [not found in any Ubuntu package repos](https://packages.ubuntu.com/search?suite=default&section=all&arch=any&keywords=uranium&searchon=names). Any possible fixes how to install Cura on Ubuntu _other_ than static app-image and thus having typical updates we've got used to?

thopiekar, 2017-07-05:
Uranium is part of my package, see:
https://launchpad.net/~thopiekar/+archive/ubuntu/cura

I guess the error message is just misleading.

isbm, 2017-07-05:
@thopiekar yep. You were right. Worked fine. The procedure is so:
```
ppa-purge ppa:thopiekar/cura
```
...lots of updates/downgrades/removes/reverts. Then:

```
sudo add-apt-repository ppa:thopiekar/cura
```

After 2.6.1 works fine. OK, this issue can be closed, thanks! Maybe just update your docs somewhere for RPM guys, who does not know that for every package in Debian world there is another tool to manage it. :laughing:

closed 2017-07-23 by BagelOrb