Ultimaker/Cura #2074: RFE: concentric zigzag infill pattern

Andrei-Pozolotin, 2017-07-05:
this request is to add concentric zigzag infill pattern, for example:
[image]

Andrei-Pozolotin, 2017-07-05:
problem with current concentric: it collapses on high print speeds
problem with current zigzag: it is not concentric (works on high speeds)

BagelOrb, 2017-07-06:
Well, this pattern doesn't allow for high print speeds because the line segments are so short that the actual sped will be limited by your acceleration / jerk settings.

I'm sorry to say that I don't think this is worth investing our precious implementation time.
There are loads of cool features still on our TODO list, and the TODO list keeps overflowing.

closed 2017-07-06 by BagelOrb