Ultimaker/Cura #2076: Printer profiles deleted by Cura

Thijxx, 2017-07-06:
Today I started Cura and all my printer profiles are gone. I did not upgrade Cura, I did not reboot my computer or did any other thing with disks or permissions. I found a cura folder in `~/.config` and all it contains is a `cura.cfg` file. It looks like Cura selfdestructed and deleted all my data..

Cura is the first software I ever encountered to **not** show the version in the _about_ window, so I do not know what the version is I am running.

When I do Help -> Show Configuration Folder all I get is an empty directory.

nallath, 2017-07-06:
Have a look in ~/.local/share/cura 

Could you share the cura.log file that's there?

Thijxx, 2017-07-06:
Here you go, renamed to **.txt** because of GitHub rules.

[cura.txt](https://github.com/Ultimaker/Cura/files/1127131/cura.txt)

Ghostkeeper, 2017-07-07:
This sort of looks like the log of a run from source code, since libSavitar and zeroconf can't be found and as far as I know that has always been built fine in our public releases.

It also looks like the layer height ended up at 0 at some point, which then caused all profiles to crash due to division by 0. Normally I'd just expect Cura to load the profiles though and indicate after starting up that the layer height is wrong. At least that's what happens for me (I just tested it).

ChrisTerBeke, 2018-07-30:
No response for a loooong time. Closing.

closed 2018-07-30 by ChrisTerBeke