Ultimaker/Cura #2077: Default Material profiles between Windows and OSX version are different

Careless, 2017-07-06:
I installed the windows and the macOS version of Cura 2.6.1 on both my computers. I was attempting to share material profiles between the two operating systems, but noticed that a fresh install on both systems does not have the same default material profiles.
- [profile comparison photo](https://i.imgur.com/bxLdfRB.png)

Furthermore, I could not import my material from Windows to macOS. It displays a success message, but refuses to add the material to the list.
- [success... but no material present on macOS import](https://i.imgur.com/xhl6bZ0.png)

I deleted and imported the profile on the Windows version and it imported without issue.
- [success on windows import](https://i.imgur.com/PQqOf4L.png)

Printer is the included Folger Tech FT-5, no changes to the printer profile other than the name changed to "FolgerTech" because I like the way it looks without the space... tried with unmodified name as well, same thing.

NOTE: importing quality/printing profiles from one operating system to the other DOES work, just not materials.

Careless, 2017-07-06:
on Windows, I deleted the configuration folder, uninstalled Cura, and then reinstalled and ran Cura. 

I added the printer, checked materials, and now it's the same as the macOS list with just Ultimaker profiles.

[and now it also does not import my PLA profile into the list, yet provides a success message.
](https://i.imgur.com/4DKUrAX.png)

checking the configuration folder on both Windows and macOS, it is [renaming the material xml file.
](https://i.imgur.com/0x0xCDt.png)

bug?

fieldOfView, 2017-07-06:
You seem to have different material diameters on Windows and Mac OS

Careless, 2017-07-06:
well how do i get the 1.75mm filament profiles to show up, because the FolgerTech FT-5 is not a 2.85mm printer out of the box, so the profile should be 1.75mm.

I guess I have to copy all the data from the available profile into a custom profile? in order to use the same settings but have the right filament show up?

perhaps upon importing a filament that has a 1.75mm diameter, the program should tell the user "you currently have a machine with a 2.85mm diameter filament as your activated device".

would make the UI a lot more friendly.

fieldOfView, 2017-07-06:
The quickest way is to set the material diameter in the sidebar once. If the material diameter is not currently visible in the sidebar, just type "diameter" in the "search..." box. 

Yes, there's still some work left.

Careless, 2017-07-06:
Is there ever going to be a way to just export all settings within the program to an xml file of some sort and import it onto another computer/OS?

Would be great if I can just do a pull of a config file from my server if I ever need to, or even have a shared network profile.

fieldOfView, 2017-07-06:
I don't know about "ever", but currently there is no such one _file_. There is one _folder_ though:

Help -> Show configuration folder...

You could symlink that folder into something that is shared/networked. Note that having the configuration open in multiple instances may lead to some data getting overwritten/lost.

Careless, 2017-07-06:
I've already attempted to copy that entire folder from Windows to macOS and vise-versa. The program crashes and requires that I remove all the contents to get it to start again.

I've also attempted to just copy the visible settings portion of the cura.cfg file so that the interface is at least identical, and it just ignores it on the windows cfg file.

Not having much luck here. makes working on two or three computers quite a chore- especially if some features are hidden on one computer and declared visible on another.

Ghostkeeper, 2017-07-09:
Lots of varied things being said here... In short, what I think happened is that you accidentally set the material diameter to 2.85 on MacOS. It's a normal setting now and could be overwritten by any quality/custom profile you have, or just as current changes.

> but noticed that a fresh install on both systems does not have the same default material profiles.

There should be no difference between Windows and MacOS in the way it stores and uses profiles. There is only one source file for the Folgertech FT-5: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/folgertech_FT-5.def.json . It hasn't been modified since its first introduction in February. Perhaps you manually changed the material diameter? What does your Diameter setting say under MacOS (under the Material category)? What happens when you change that to 1.75?

> Furthermore, I could not import my material from Windows to macOS. It displays a success message, but refuses to add the material to the list.

> perhaps upon importing a filament that has a 1.75mm diameter, the program should tell the user "you currently have a machine with a 2.85mm diameter filament as your activated device".
>
> would make the UI a lot more friendly.

The material file probably specifies a filament diameter of 1.75, and since your printer on MacOS somehow ended up at 2.85, it filters them out. I agree that it's confusing that no message appears to warn you.

> on Windows, I deleted the configuration folder, uninstalled Cura, and then reinstalled and ran Cura.

You might also want to delete the cache of printer definitions in `C:\Users\<Careless>\AppData\Local\cura`. It shouldn't be a problem with a clean install, but you never know with this sort of problems.

> checking the configuration folder on both Windows and macOS, it is renaming the material xml file.

Most likely not a bug, but it's just renaming the file because a file with the same name already exists. The `%23` stands for the `#` character.

> Is there ever going to be a way to just export all settings within the program to an xml file of some sort and import it onto another computer/OS?

Product management is looking into a different type of solution to this problem, but I don't think I can disclose those at this point. Nothing is set in stone except hieroglyphics.

> I've already attempted to copy that entire folder from Windows to macOS and vise-versa. The program crashes and requires that I remove all the contents to get it to start again.

I would love to see the log of that crash :)

> I've also attempted to just copy the visible settings portion of the cura.cfg file so that the interface is at least identical, and it just ignores it on the windows cfg file.

Make sure to close Cura first before you copy it, otherwise Cura will just overwrite the copied file again on the point that it closes.

Ghostkeeper, 2017-07-09:
I've made an internal ticket to make a message display when a material profile is not supported by the current set-up.

Careless, 2017-07-09:
I'll see if I can get a log when I have time to test the transfer of the config folder from Windows to macOS again.

But I never changed the material to 2.85. I would never have any such reason to, as I do not have 2.85mm filament at all. I don't know why the program ever started displaying 2.85mm filament profiles- which is why I made this post. I didn't know why the filament list was different until @fieldOfView informed me of the filament diameter being the root cause. How it got to that point, I do not know. I only ever selected the FolgerTech FT-5 profile as I had no reason to select any other.

Maybe it was a setting left over from a previous version of Cura? Though I can 100% with absolutely certainty tell you that I never actually used Cura or sliced a file and printed it on my printer, so if being a setting left over from a previous version were true, it would have had to have been from Cura 15.x which was long ago when I was just testing out the interface. Even then, I couldn't tell you why the filament was set to 2.85mm because I'd have had no reason to do so back then either.

I just recently started slicing and actually printing files generated by Cura, and in the most recent install of 2.6 and 2.6.1, I never selected 2.85mm. And that's all I can tell you about that as far as my printer, filament, and settings go because that's all the information I have regarding the installation of the program.

Regardless, lets say a user does select 2.85mm via a custom profile or something. And then Chooses to use 1.75mm filament, some questions I have are:

- why isn't the list updated to show the 1.75mm material profiles
- why isn't there a tabbed window to list 2.85mm filaments and 1.75mm filaments so the user doesn't have to choose one at the outset of selecting a printer, and have the list only populated with those materials?

It's like you have to start over if you decide to change diameters, which doesn't make sense since the user can't REMOVE the filaments that are populated on the list.

If I started at 2.85mm somehow, created a bunch of 1.75mm profiles, I'd have to export them all, and then reimport them into a newly created machine profile that started out at 1.75mm.

The whole material list being one size or the other doesn't quite make sense to me. I think the diameter thing could be integrated into the UI a bit better than how it's currently done.

Ghostkeeper, 2017-07-10:
We don't automatically import settings from Cura 15.x. The first version to auto-upgrade settings was 2.3 and it only understood 2.1 settings (sorta). Since the FolgerTech printer never changed since its publication, I don't think it was leftover from a previous installation.

Perhaps it managed to select a 2.85mm material once for some reason. Perhaps as a fallback because the material files were bugged? Once it does so, the material diameter would change and consequently the list of materials would change.

> why isn't the list updated to show the 1.75mm material profiles

I expect the list would get updated, actually. I just tried and that seems to be working properly.

> why isn't there a tabbed window to list 2.85mm filaments and 1.75mm filaments so the user doesn't have to choose one at the outset of selecting a printer, and have the list only populated with those materials?

We like to hide things a bit better in this case. It's very rare that the user ever wants to put 2.85mm filament in their printer of which we know that it only supports 1.75mm, so we don't waste any interface real estate to allow the user to do so. If an expert user wants to do that anyway, he can adjust the material diameter in the material settings.

> If I started at 2.85mm somehow, created a bunch of 1.75mm profiles, I'd have to export them all, and then reimport them into a newly created machine profile that started out at 1.75mm if I wanted the pre-populated profiles and custom profiles on a list together.

There was a bug in 2.6 also where updating the material diameter in the material manager (rather than the sidebar) would not cause the material list to update. That was fixed in 2.7, and at the same time a dialogue box was added that warns you when a material disappeared from the list due to the diameter.

The UI is confusing to a lot of people. If we can come up with a better solution that still hides the materials that can't be used in your printer, we would love to get that in.

Careless, 2017-07-10:
> The UI is confusing to a lot of people. If we can come up with a better solution that still hides the materials that can't be used in your printer, we would love to get that in.

May I suggest a "Only show materials of currently selected diameter" checkbox under the main listing for materials, or a "Hide materials of other diameters"

> I expect the list would get updated, actually. I just tried and that seems to be working properly. 

> If I started at 2.85mm somehow, created a bunch of 1.75mm profiles, I'd have to export them all, and then reimport them into a newly created machine profile that started out at 1.75mm if I wanted the pre-populated profiles and custom profiles on a list together.

I guess I may have changed it at the wrong spot in the app. Which begs the question- why are there two areas to change it which have different overall effects...

> There was a bug in 2.6 also where updating the material diameter in the material manager (rather than the sidebar) would not cause the material list to update. That was fixed in 2.7, and at the same [...]

Where does one test 2.7? I only see 2.6.2 available.

Ghostkeeper, 2017-07-10:
The 2.7 beta is planned to be released in 3 or 4 weeks.

ChrisTerBeke, 2018-01-22:
Is this still an issue? Or was it due to material diameter differences?

nallath, 2019-04-04:
Closing this issue since it's quite old and unlikely to be still be an issue.

closed 2019-04-04 by nallath