Ultimaker/Cura #2078: Feature request: Cloud Sync

kr15uk, 2017-07-06:
Online account sign in to sync for all your Machines, Profiles and Materials.
GUI customisation (like Settings and General) sync would be nice but could live without it.

nallath, 2017-07-06:
A first step has been set with regards to putting some of Cura's data remotely. The 2.7 release will have a "plugin browser", which allows you to add new functionality to Cura. 

Depending on how well that goes, we will look into other cloud based features. I think that putting a number of machines / materials / profiles remote (so they are only downloaded as needed) will go quite a way for the speed of Cura.

ChrisTerBeke, 2017-11-14:
Duplicate of https://github.com/Ultimaker/Cura/issues/1310

closed 2017-11-14 by ChrisTerBeke