Ultimaker/Cura #2084: No File Nmae

Waggy1, 2017-07-07:
If I open Cura and then find the object I want to print, the output has the same name as the object file.  If I click on the object file and that causes Cura to startup and then load the file, there is no file name.  I just get .GCODE where I should have filename.gcode

Ghostkeeper, 2017-07-10:
What version are you using? This was a known issue but it was fixed for the 2.6.0 release.

Waggy1, 2017-07-10:
I am using 2.5.0.  Is it your software that causes the top layer not being filled in?  I just made a door stop that I downloaded that had the word stop making the stop.  Just before the top there were a few holes, like the fill was not covered.  Then I made a U bracket and the top section of the bottom of the U has holes.  What program determines how thick the top layer over the fill is going to be?  I have attached a picture with pens pointing to the problems.

 

Thank You

Wayne

nallath, 2017-07-10:
The auto email reply of github doesn't handle attachments. Could you add a link to the image?

Ghostkeeper, 2017-07-10:
The thickness of the layer of material that goes over the infill is a setting called "Top Thickness". You can find it by typing that in the search bar above the settings list.

ianpaschal, 2018-02-27:
Needed more info and was never provided, so closing.

closed 2018-02-27 by ianpaschal