Ultimaker/Cura #2085: Abort does not reset extruder position

Thijxx, 2017-07-08:
It would be extra nice if the extruder was moved to a 'safe' location when a print job is aborted. Now it's just frozen and I always need to move the axis manually so I can remove the failed print and clean the nozzle.

Ghostkeeper, 2017-07-10:
This is during USB printing I assume.

When printing via the network with an Ultimaker 3, the abort procedure of the printer is called which homes the nozzle by itself. I don't know exactly what happens when printing via OctoPrint.

With USB printing, I think we can fix this with a simple added "G28" command to home the nozzle. I'll try to take a small look into it tomorrow.

Ghostkeeper, 2017-07-10:
Fixed here: https://github.com/Ultimaker/Cura/commit/6d2b46244f910cc5f09b07bdde1955bdcde77afb

Thijxx, 2017-07-10:
Awesome! I should have mentioned that I use a **Prusa i3 MK2S** and that I print over USB, sry bout that.

closed 2017-07-10 by Thijxx