Ultimaker/Cura #2087: Cura 2.6.2 4k scaling

shahna0c, 2017-07-08:
Hello, 

I upgraded Cura software to 2.6.2 for better features. However, the interface is not scaled properly for 4K monitor (the layer selection bar is disappeared). 

Anyone can help to overcome this issue? Your support is highly appreciated. 

Thank you
[image]

Ghostkeeper, 2017-07-10:
Wow, it made that sidebar like 1700 pixels wide for some reason.

We've been having some problems with Windows 10 changing their DPI scaling like every month or so. I think it's synced to the menstrual cycle of Nadella's wife. And it's not really up to us, but up to the library we're using to display these things, Qt. And partially up to us because we've had some problems with improperly using said library sometimes...

Ghostkeeper, 2017-07-10:
In the 2.6.2 build we reverted to an older Qt version. Perhaps you could test with 2.6.1 to see if anything changed? If 2.6.1 is working better, then it might also be working better in 2.7.

fieldOfView, 2017-07-10:
Note that this issue will be mitigated somewhat in Cura 2.7 by having the layerview panel move up when it does not fit below the viewmode button.

awhiemstra, 2017-07-10:
From the looks of it, it seems to use about double your font size for the default font, rather than the same font size as the title bar. Since everything is sized relative to the default font size, you end up with quite a large UI. Maybe there is a workaround where you can tweak the default font size?

ianpaschal, 2018-02-27:
This is fixed in version 3.2.1.

closed 2018-02-27 by ianpaschal