Ultimaker/Cura #2089: Reopen  previous files

MiguelAngelLV, 2017-07-09:
When I open a file after open other file, Cura add again all previous files.

Example:

I open file 1.stl, then, open file 2.stl. In this moment, cura add 2.stl and 1.stl again. After I open file 3.stl, so cura add 3.stl and 2.stl and 1.stl again... So I have 3 1.stl, 2 2.stl and 1 3.stl.

Ghostkeeper, 2017-07-10:
I suspect it's your operating system doing this. We retain no memory of previous files other than what's in the "File->Open Recent" menu, and we certainly don't auto-load those.

fieldOfView, 2017-07-10:
If I remember correctly, this behavior on some Linux flavors was the reason why opening multiple files at the same time was disabled until Cura 2.4 (?). @awhiemstra, do you remember?

MiguelAngelLV, 2017-07-10:
With "open recent" works fine. So, is the "Open File Menu" the problem.

nallath, 2017-07-10:
It was only with kubuntu (if I recall correctly) that we removed multi file loading.

awhiemstra, 2017-07-10:
It is a bug in the KDE version of the Open File dialog. See https://bugs.kde.org/show_bug.cgi?id=350408

MiguelAngelLV, 2017-07-10:
I use KDE, so, it would be the problem.

Ghostkeeper, 2017-07-13:
We have two options then:
* Use a different Open File dialogue on KDE as well. The bug report awhiemstra mentioned is only reported with multi-file dialogues so perhaps a single-file dialogue works fine.
* Say that it's up to KDE to fix this and do nothing.

I'm voting for the first: A workaround for KDE only.

Ghostkeeper, 2017-10-10:
Now it's become the second. Since this ticket wasn't touched for 12 weeks, our project manager removed it from our planning.

closed 2018-02-27 by ianpaschal