Ultimaker/Cura #2090: Cura 2.6.1: Default Prime tower position invalid

Dim3nsioneer, 2017-07-09:
The situation this issue occurs is when you have two materials with very different settings for the brim size, e.g. pp (very large brim) and tpu-95a.
To reproduce the error do this:
- load two objects
- select pp in first printcore, eg. tpu-95a in second printcore
- assign one object to the second extruder
It seems Cura places the prime tower according the settings of printcore 2 which happens to be in the forbidden range of printcore 1.
[image]

Ghostkeeper, 2017-07-10:
Yeah, we have a bit of a problem here. We keep moving that prime tower ever so slightly further towards the centre because we keep getting profiles with large brims or rafts.

In this case, the PP profile is the culprit. We'll have to move the prime tower again, but that's no permanent solution. Perhaps we could devise a way to place the prime tower always just inside with an inheritance formula... I might give that a try tomorrow.

Dim3nsioneer, 2017-07-10:
Cool, thx.

Ghostkeeper, 2017-07-10:
Fixed this for the Ultimaker 3 here: https://github.com/Ultimaker/Cura/commit/ee52e1b9eaf055433ce5c3367045402c078b50df

closed 2017-07-10 by Ghostkeeper

Dim3nsioneer, 2017-07-10:
I guess you are aware that at a certain brim size the x coordinate still can cause the same issue?

Ghostkeeper, 2017-07-13:
Yeah, I'm aware, because of the bed clip and such. We're not running into the limit right now, so I think I'll leave that for later :)