Ultimaker/Cura #2093: i can't open modify g-code gui with cura 2.3 2.6

cisforcer, 2017-07-10:
i am using cura 2.3/2.6  64 with win7 64x
anyone know what happen to me？

cisforcer, 2017-07-10:
this is 15.04.6 “pause at z“   modified code ，i insert two pause but   its show in one layer

;LAYER:47
;TYPE:CUSTOM
M83
;TYPE:CUSTOM
M83
G1 E-5.000000 F6000
G1 X111.100000 Y11.100000 F9000
M84 E0
M0
G1 E5.000000 F6000
G1 E-5.000000 F6000
G1 X90.829000 Y94.506000 F9000
G1 E5.000000 F6000
G1 F9000
M82
G1 E-5.000000 F6000
G1 Z15 F300
G1 X190.000000 Y190.000000 F9000
M84 E0
M0
G1 E5.000000 F6000
G1 E-5.000000 F6000
G1 X90.054000 Y93.793000 Z5.000000 F9000
G1 E5.000000 F6000
G1 F9000
M82
G0 F9000 X90.054 Y93.793 Z5.000
;TYPE:FILL
G1 F2580 X96.496 Y87.351 E90.13785

Ghostkeeper, 2017-07-10:
It's not under Extensions -> Post Processing -> Modify G-Code?

What happens when you click on that in the application menu?

cisforcer, 2017-07-10:
yes its there but click no reaction

cisforcer, 2017-07-10:
no color change when mouse move on

Ghostkeeper, 2017-07-10:
Perhaps you could share a fragment of your log then?

The log will be located under `C:\Users\<cisforcer>\AppData\roaming\cura\cura.log`.

cisforcer, 2017-07-10:
2017-07-10 08:17:47,114 - DEBUG - CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [474]: Slicing took 1.928110122680664 seconds
2017-07-10 08:17:47,117 - DEBUG - UM.Backend.Backend._logSocketState [176]: Socket state changed to Closing
2017-07-10 08:17:47,119 - DEBUG - UM.Backend.Backend._onSocketError [198]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2017-07-10 08:17:47,120 - DEBUG - UM.Backend.Backend._logSocketState [178]: Socket state changed to Closed
2017-07-10 08:17:47,121 - DEBUG - CuraEngineBackend.CuraEngineBackend._onBackendQuit [585]: Backend quit with return code 0. Resetting process and socket.
2017-07-10 08:17:47,654 - ERROR - SliceInfoPlugin.SliceInfo.run [58]: An HTTP error occurred while trying to send slice information: HTTP Error 500: Internal Server Error
2017-07-10 08:17:52,616 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-07-10 08:19:00,847 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-07-10 08:23:53,968 - DEBUG - UM.Controller.setActiveView [85]: Setting active view to LayerView
2017-07-10 08:23:54,236 - DEBUG - CuraEngineBackend.ProcessSlicedLayersJob.run [232]: Processing layers took 0.2580146789550781 seconds
2017-07-10 08:24:13,271 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-07-10 08:24:18,719 - ERROR - UM.PluginRegistry._populateMetaData [323]: Could not find plugin global_stack
2017-07-10 08:24:18,723 - ERROR - UM.PluginRegistry._populateMetaData [323]: Could not find plugin extruder_stack
2017-07-10 08:24:35,489 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-07-10 08:33:22,824 - DEBUG - UM.Qt.QtApplication.windowClosed [267]: Shutting down cura
2017-07-10 08:33:22,928 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [235]: Attempting to kill the engine process

fieldOfView, 2017-07-10:
@cisforcer, can you try if you can move the focus in the menu with your keyboard (cursor keys) and pressing enter when the right menu entry is focussed? I remember this being reported once or twice before, but hard to find a computer that reproduces the issue.

cisforcer, 2017-07-10:
let me  try now

cisforcer, 2017-07-10:
@fieldOfView  yeah!!!!!!!   i open the modify GCODe  with keyboard~~ but no raaction with mouse

cisforcer, 2017-07-10:
[image]
[image]

cisforcer, 2017-07-11:
@fieldOfView  thank you very much~~~~ and the new pause at z working good.

closed 2017-07-13 by cisforcer