Ultimaker/Cura #2095: CURA 2.6 no solution to deconnect port COM

spilz87, 2017-07-10:
Hello,

with CURA 2.6, when a printer is connected through USB, it automatically open by CURA, and there is no way do disconnect it ??

I'm using CURA as slicer, but not to contrôle my printer 

I don't know if we can call it an "issue"

regards

fieldOfView, 2017-07-10:
You can disable the "USB printing" plugin in the Plugins pane of the preferences.

closed 2018-02-26 by ianpaschal

Ghostkeeper, 2018-03-16:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/1049.

jandrieu, 2018-08-08:
It's not a duplicate. I still have this problem in 2.4.0. There is no "X" for disabling the plugin. Yes. I've gone to the preferences and found the USB plugin. Yes, I've tried making my window wider (no change). There simply isn't a way to disable ANY plugin on my installation.

fieldOfView, 2018-08-08:
Cura 2.4 is rather old. Disabling plugins was introduced in Cura 2.5, I believe (but it was a loong time ago).

jandrieu, 2018-08-09:
Ok. Cool. I'm glad to know an upgrade will fix it.

fieldOfView, 2018-08-09:
@jandrieu just to be sure, what version of Cura are you using? Are you sure 2.4.0 was not a typo? Cura 3.4.1 is the current version. In Cura 3.4.1, you can disable the USB printing plugin via the "Toolbox" on the  "installed" tab.

jandrieu, 2018-08-09:
Not a typo. I installed it when I bought my printer a couple years back and, since it mostly works great, haven't updated.

Mr-What, 2019-08-18:
not finding this on 4.2
no plugins in preferences

fieldOfView, 2019-08-18:
It is in the Marketplace now; there’s an “Installed” tab that has the functionality that used to be in the preferences.