Ultimaker/Cura #2097: cura-plugin-orientation crashes cura

maukcc, 2017-07-10:
os ubuntu 16.04
Cura 2.6.1 stable

cura-plugin-orientation crashes cura

fieldOfView, 2017-07-10:
[image]?

maukcc, 2017-07-10:
:) 
When we install cura-plugin-orientation, Cura will not start.
When we remove cura-plugin-orientation, Cura will
[cura.log.txt](https://github.com/Ultimaker/Cura/files/1135306/cura.log.txt)
 start.

nallath, 2017-07-10:
You didn't use the right plugin version :( You're either using the 2.6 plugin with master or the master plugin with 2.6

fieldOfView, 2017-07-10:
> You're either using the 2.6 plugin with master or the master plugin with 2.6

If Cura crashes because of that, that is going to be an "interesting" issue when people upgrade to 2.7 in a month or so when they have plugins in their configuration folder.

nallath, 2017-07-10:
Well, it's actually using the 2.7 version plugin with the 2.6 release. There is an issue with how the old code handles invalid data; It raises an exception and stops loading other plugins.

ChrisTerBeke, 2018-07-30:
Plugin incompatibility, plus it's a plugin issue, not a Cura issue, closing...

closed 2018-07-30 by ChrisTerBeke

fieldOfView, 2018-07-30:
The underlying issue that plugins can cause Cura not to load is not fixed though