Ultimaker/Cura #2099: cura 2.6 and elementary os

mage1961, 2017-07-10:
I'm having trouble installing cura 2.6 into elementary os. does anyone have any ideas. I cant seem to get it to recognize the app imge

Ghostkeeper, 2017-07-14:
Did you do `chmod +x Cura.appimage`?

Ghostkeeper, 2017-07-14:
There's a bug report that AppImage doesn't work on Elementary OS: https://github.com/RicinApp/Ricin/issues/127

But it's over a year old, so that would have been fixed by now you'd think.

mage1961, 2017-07-14:
I'm gonna have to re try it....will try it this afternoon

peteruithoven, 2017-12-14:
Issue RicinApp/Ricin#127 seems to be about Elementary OS 0.3 Freya (based on Ubuntu 14.4) (released April 2014), in the mean time Elementary OS 0.4 Loki (based on Ubuntu 16.04) was released quite a while back on 9 September 2016. I've been using Cura 3 and 3.1 on the latest Elementary OS Loki and it works fine. So maybe this issue can be closed? 
Issue RicinApp/Ricin#127 itself is also closed.

Ghostkeeper, 2018-01-02:
All right, thanks for retesting.

closed 2018-01-02 by Ghostkeeper