Ultimaker/Cura #2100: Non smooth surface texture when printing cylinder

quebeck5, 2017-07-11:
Using Cura 2.6.2

The surface texture when printing a cylinder is not smooth at certain heights. 

Attached photo shows the cura print on the right, and the same print on the left with slic3r with same settings.

Attached zip file has the STL of the cyclinder, and the gcode output from Cura and Slic3r.

[cura cylinder bug.zip](https://github.com/Ultimaker/Cura/files/1137659/cura.cylinder.bug.zip)

[image]

BagelOrb, 2017-07-13:
Are you sure you attached the cura cylinder gcode?

The test_cylinder.gcode looks perfectly fine to me. There's no vibrating moves in there.

Ghostkeeper, 2017-07-14:
Well it says `;Generated with Cura_SteamEngine 2.6.2` at the top of the file...

Ghostkeeper, 2017-07-14:
The file from Slic3r prints at at `F1380`. The file from Cura prints at `F2160`. I think you just need to reduce the speed in Cura a bit.

closed 2017-07-14 by Ghostkeeper