Ultimaker/Cura #2101: can't  open  cura 2.3 （32） with old computer xp system

cisforcer, 2017-07-11:
and   cant find any  log files  to show  what happen .when i click the cura.exe   nothing happen ,nothing show

fieldOfView, 2017-07-11:
I don't think Windows XP is supported, not even by Cura 2.3.

cisforcer, 2017-07-11:
thank you very much

rmcolbert, 2017-07-12:
The folder paths used in Cura for logging and user data didn't exist in XP so it may be crashing trying to create the logs. Download the SysInternals application FileMon or ProcMon and run that before launching Cura. You should be able to see where Cura is attempting to write files to get (at the very least) the error log.

Ghostkeeper, 2017-07-14:
Yes, we require Windows 7 or higher from our Windows users.

Or rather, we require Qt 5.7 which requires Windows 7.

The crash happens before we even try to log anything.

closed 2017-07-14 by Ghostkeeper