Ultimaker/Cura #2102: All Ultimaker profiles randomly deleted but selected

cmatthes, 2017-07-11:
Yesterday I reconnected to my Ultimaker 3 with Cura 2.6.2 on my mac (MacOS 10.11.6) and choose not to sync with the materials loaded. All the sudden every ultimaker profile disappeared except the currently selected. 
Adding a new printer didn't resolve the issue. 
Needed to reinstall Cura to get the profiles back. This was the second time this happened to me running the Cura Version 2.5+ 
Attached the log as well as some screenshots.
[image]

[image]

[cura.txt](https://github.com/Ultimaker/Cura/files/1138270/cura.txt)

Kilroy07, 2017-07-11:
I second this, everything was fine yesterday. Today I try to slice a few models and Gcode is WACKED!! Looked more into it and all my custom profiles, materials etc are ignored!

ADAPT-Shop-Brady, 2017-07-11:
Same with me.... Also have filament in printer that can't be chosen on lcd so I choose not to update cura with printers settings (T-Glase loaded, chose PLA when loading it).

nallath, 2017-07-13:
Could you share your logs & project file of current state?

cmatthes, 2017-07-13:
I attached the log file to my original posting (cura.txt). 
What specific project file are you for?

nallath, 2017-07-13:
Just save a project if the state of cura is like that.

cmatthes, 2017-07-14:
Attached the requested project file as well as the 2.6 folder from application support.
[profiles-deleted.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/1148324/profiles-deleted.curaproject.3mf.zip)
[2.6.zip](https://github.com/Ultimaker/Cura/files/1148325/2.6.zip)

nallath, 2017-07-17:
You changed the material type. This is used by Cura to match a profile. As it doesn't know "support" (eg; it has no profiles to match it with), it shows an empty list.

closed 2017-07-17 by nallath

cmatthes, 2017-07-22:
Thank you @nallath for looking into it and explaining what I can do to avoid it! :) 

But Shouldn't Cura not show any profile at all if it a material you don't know about? 

> We're looking into improving this behavior, but as of now, we can't support all combinations (especially if it's a material we don't know about).

Looking forward to it ! 
Please have a look at how Kisslicer handles material, machine etc. profiles. IMHO it's the right approach to separate and group not related and connected settings. 
Maybe Cura does it in some kind of way. However the UI makes it kinda hard to notice and understand.

ZeroScars, 2017-08-03:
@nallath 
As users derive a material from one of the existing you know what material is being used, let that material couple to the print profiles the user creates him or herself, now when you create and save a print profile for a custom material it isn't even listed for that custom material! That should not be to difficult to fix I guess. It looks as though the material and print profile process is a little too complex to work with. There are many more wonderful materials that Ultimaker itself does not sell but would love to be printed on the UM3.

Ghostkeeper, 2017-08-20:
I do consider that a bug you got there, ZeroScars... Reproduces for me in 2.7.

ZeroScars, 2017-08-22:
@Ghostkeeper

Strange thing is that when you created new profiles for a new derived material, the profiles do list up in the parent material (the one you derived your custom material from).

The ability to have and use custom profiles for custom materials would be a bless. I print a lot of non-Ultimaker materials (besides the Ultimaker PLA, nylon, PVA) like PETG and POM on my UM3, but I have to change a lot of settings each time.

Should a new ticket be created, this one is closed unfortunately.

Thanks, Oscar

Ghostkeeper, 2017-11-23:
That second bug is a duplicate of #1689.