Ultimaker/Cura #2104: machine_extruder_start_pos

maukcc, 2017-07-12:
Can we make it so that:

> machine_extruder_start_pos_x

and

> machine_extruder_start_pos_y

are used on every tool change.
And not only on the 1st layer when prime_blob_enable is enabled.

Ghostkeeper, 2017-07-17:
We do though! It's right here: https://github.com/Ultimaker/CuraEngine/blob/3b8eb11765e07fd2c6a9d5021eaf45d688e5014a/src/LayerPlan.cpp#L212

The last_planned_position gets updated at every extruder switch. We use it to set the time estimates (and consequently for other things such as pre-heating the nozzles).

maukcc, 2017-08-15:
Yes, what we want is for the extruder to explicitly move there after a tool switch.
This actually happens on the 1st layer when prime_blob_enable is enabled.

I am guessing that the values of the machine_extruder_start_pos_x and y are updated every layer with positions close to where it should be starting with printing the object?

But then the machine_extruder_start_pos_abs set to true makes no sense, as it is not absolute but object related.

Ghostkeeper, 2017-08-29:
The settings are meant to indicate where the _firmware_ puts the nozzle after a switch, so we can properly avoid objects when moving back. I don't think we should repurpose them for any other functionality.

What you seem to need is sort of like a nozzle wipe position (to Cartesio's printhead cleaner?) where it goes after every switch additionally.

Ghostkeeper, 2017-08-29:
You could implement a nozzle wipe in the `machine_extruder_start_code` setting.

maukcc, 2017-08-29:
We tried that many months ago😀
The problem is that the machine_extruder_start_code is placed before the extruder heatup (M109) command.
So then the nozzle would get wiped, and then sits there waiting for the temp to be reached and all that time its oozing again.

maukcc, 2017-08-29:
So what does machine_extruder_start_pos_abs do?

Ghostkeeper, 2017-09-03:
It makes the move to `G0 X{machine_extruder_start_pos_x} Y{machine_extruder_start_pos_y}` a relative move if disabled. The most important function of this is that you can then set the start position to 0, 0 relative, meaning that it won't move to the start position at all. You could also use it to move the nozzle slightly away from the wall or something, but there is no guarantee that there is not another wall in the new position.

closed 2018-02-09 by maukcc