Ultimaker/Cura #2111: wrony layer when  use pause at height

cisforcer, 2017-07-13:
i seted pause at 13mm but 5mm in code

cisforcer, 2017-07-13:
maybe i click to fast？

nallath, 2017-07-13:
I just checked this, it seems to work for me. Could you check if this always happens or if it was a one time issue?

cisforcer, 2017-07-17:
if i add to more than  2 plugins and changing the setting fast it will  happen. if  i change settings after slicing  everytime ， it will not happen.

fieldOfView, 2017-07-17:
@Nallath: sounds like another case of the activeFocus needing a tickle

Ghostkeeper, 2017-07-17:
Maybe it's related to that bug where it doesn't auto-reslice when you change a setting just when it's done slicing.

cisforcer, 2017-07-19:
maybe  put a bottom in “add a script”  that  we can slice manually.

Ghostkeeper, 2017-07-19:
It's not the focus. At least, when I change the Z-position setting and click "Close" without moving the focus away from the checkbox, then save the g-code to a file, the pause occurs at the correct height.

cisforcer, 2017-07-19:
i changed the setting  before the auto slicing start，the pause code would be correct.

fieldOfView, 2017-07-19:
@Ghostkeeper: that shows the activeFocus is correctly handled before closing the window. What @cisforcer mentions about 2 or more plugins (here: https://github.com/Ultimaker/Cura/issues/2111#issuecomment-315650489) suggests the activeFocus may not be correctly handled when selecting another script from the listview.

As a sidenote: there is no reason why changing a postprocessing script should trigger a reslice. Scripts are executed when saving the gcode, they do not affect slicing or the sliced result at all. They only affect the saved (or sent) gcode.

Ghostkeeper, 2017-07-19:
Yes, but it's still necessary to reslice in some cases because once it's been post-processed we save the processed g-code in the scene root. When you save the file to two locations it doesn't process the g-code twice.

Of course, if the g-code is not post-processed yet you wouldn't have to reslice... But in some cases you do have to reslice.

cisforcer, 2017-07-23:
I did the modify  of  the  scripts file and slic slowly  and its working good

Ghostkeeper, 2017-07-24:
We haven't been able to reproduce this yet.