Ultimaker/Cura #2112: code for support structure generation

gokul2907, 2017-07-13:
Can someone help me in finding the code for support structure generation (angle between building direction and facet normal)

nickthetait, 2017-07-13:
Start looking here https://github.com/Ultimaker/CuraEngine/blob/master/src/support.cpp

BagelOrb, 2017-07-13:
The facet normal isn't used. Instead we compare polygons with offsets equal to the tangent of the max support angle multiplied by the layer height.

gokul2907, 2017-07-13:
Thanks for your reply.I will slightly rephrase my question, how to identify the areas in CAD  that required support structure (need code for this)

BagelOrb, 2017-07-13:
I cannot give you code for your cad program because I'm not a software
engineer for your cad program.

However, if you can code yourself then I can give you the following hint:
compare the z component of the normalized normal vector of each face with
the tangent of your max overhang angle.

Also note that in order to do it perfect you need to check for vertices
which are connected only to higher vertices and you need to check edges for
being overhanging by looking at the z component of the normalized edge.

Good luck!

BagelOrb, 2017-07-14:
I suppose this answers your question.

closed 2017-07-14 by BagelOrb