Ultimaker/Cura #2113: Questions for making a useful json profile in Cura

RNDAlex, 2017-07-14:
I having a DIY machine and a row CuraEngine(but I had read the source code and know something slicing details). Cura is useful for me but I still want to make my own slice-software because Cura is too heavy.
My questions are:
1、I noticed that different json files depend on what machine type user chosen were used in Cura and make some parameter limited. Like default/maximum/minimum written in profile. How does Cura use these to set the limit for setting blank?
2、 Some values of setting effected by other parameters. The statement like "IF...ELSE" written in profile and How does Cura parse these meaningful string? Or it just a normal text? 
In fact, I want to know how to make a good json file to add my own machine to CuraEngine, do some cool stuff for my own software.

nallath, 2017-07-14:
Cura never has a "blank" setting. The min & max values indicate in what range the value can be. The engine is not expected to handle values outside of this range. The warning ranges are sanity checks; We think it won't give good results, but it might be possible for it to work.

The value property is a python script that can be evaluated by the frontend. 

CuraEngine doesn't have machines, all the machines (and parsing, inheritance, profiles) "live" in the frontend.

RNDAlex, 2017-07-14:
@nallath Thx, "python script" is the answer. The description of my questions may not very clear. What make s me confused was how to handle the data from profile and then "put them" to the text box for limiting the user input.

BagelOrb, 2017-07-14:
Each setting which is a number should have at least either a minimum value or a minimum warning value. The same holds for the maximum.

I suppose your questions are answered.

closed 2017-07-14 by BagelOrb