Ultimaker/Cura #2118: wont print horizontal surfaces

tcgfilm, 2017-07-15:
I cannot get this software to print horizontal surfaces!!!!
[image]

tcgfilm, 2017-07-15:
Any level surface disappears from the print.
There is supposed to be a a solid bottom and top on this arch object.

tcgfilm, 2017-07-15:
Is there anyone supporting this software other than forums?.

smartavionics, 2017-07-15:
You can open an issue on the GitHub Cura or CuraEngine repos. But before you do, please provide more information about your model and the settings you are using. Can you make the model available for people to look at? How many top/bottom layers are you using, etc. BTW, that image rather looks like it's using spiralize mode and spiralize is never going to be able to print a horizontal surface.

fieldOfView, 2017-07-15:
The image also looks like a very old version of Cura, which is no longer supported. Please use Cura 2.6.2

tcgfilm, 2017-07-17:
I'll try reloading a newer version.

Thanks,  JP.
[image]
[image]

BagelOrb, 2017-07-21:
You have the setting Only Follow Mesh Surface enabled. That setting causes the engine to only print the outlines in the horizontal direction.
If you want to print the topbottom you should disable that setting.

Perhaps you jsut wanted to set the infill peercentage to zero?

closed 2017-07-21 by BagelOrb

tcgfilm, 2017-07-21:
Thanks Tim.

That’s great to know.

I’ll look into it.

Cheers,   Joe.

Ghostkeeper, 2017-07-24:
This is a question about legacy Cura. Please direct those questions to https://github.com/daid/LegacyCura/issues