Ultimaker/Cura #2120: Lay Flat Does Not Work

slurton, 2017-07-15:
Lay flat does not work on small objects.

BagelOrb, 2017-07-21:
How small?
What kind of shape do they have?

Can you attach the file - or if that's not possible: a screenshot?

slurton, 2017-07-21:
The part was no more than 2mm in width. I discovered that if I took snap off it worked.

BagelOrb, 2017-07-21:
Problem reproduces easily with a cube scaled such that either X or Y is 2mm.

Problem goes away if total object is scaled to be larger.

This bug is genuine .

Ghostkeeper, 2017-10-30:
I also found the problem with a 10x10x2mm cuboid (Z was 2 mm, the other two dimensions were 10mm).

Ghostkeeper, 2017-10-30:
Our project manager just removed this from our planning though, because the issue is more than 12 weeks old.

closed 2018-11-22 by Appesteijn