Ultimaker/Cura #2121: Suggestion: Allow  (0)comma as well as (0)dot in model resize/scale window

Cyberman-tM, 2017-07-15:
The comma is accepted when a first number is entered, but not if the first number is missing.

So I can enter "6,6" and it'll get converted to "6.6", but I can't enter ",6" to get ".6".

Tested with Cura 2.6.2 .

Would be nice to have, german (and other?) keyboards have a comma at the numpad instead of a dot.

nallath, 2017-07-17:
0, is accepted right now. We always use the en_US locale, so we don't have issues when we're saving things (otherwise you can have problems when saving 3 models that are later loaded on a machine with a different locale).

BagelOrb, 2017-07-21:
I don't see how that's related to the fact that `0,6` is accepted, but `,6` is not.

Ghostkeeper, 2017-07-24:
There's quite a lot of problems with the DoubleValidator we're using and the hacks we've made to allow all locales. For instance, you get a crash when you enter `1,000.0`% for the scale.

ianpaschal, 2018-02-27:
@Ghostkeeper You worked on something related recently, right? Is this fixed?

Ghostkeeper, 2018-03-20:
No. You can still enter `6,6` which gets converted to `6.6`, but you can't type `,6`. You can type `.6` but Cura doesn't seem to register that.

closed 2023-06-13 by github-actions[bot]