Ultimaker/Cura #2122: Rapid move Cura

zerspaner-gerd, 2017-07-15:
I have problems printing with PETG (Colorfabb XT) with Cura (unclean pressures), with S3D I get much better print results.
Now I went on the search and you see the Cura has a lot of unnecessary travel distances.
Brief summary of recent discussion at Ultimaker Forum

### [1st Post](https://ultimaker.com/en/community/49572-cura-26-beta?page=3&sort=#reply-182459)
> When are the travel paths optimized at Cura?
> Cura has much more unnecessary travel ways, e.g. Opposite to S3D.
> This is the only thing for me, where Cura still has great weaknesses
> 
> Here example:
> [Video](https://www.youtube.com/watch?v=W9SNPaNs8-Y)
> Cura = up to 3 times in the same range, many unnecessary traverse paths
> S3D = area complete with once finished, less travel paths = for certain parts, materials better printing results

[Cura 2.5 Unnecessary travel paths video](https://www.youtube.com/watch?v=hJLYN9k2u8A&feature=youtu.be)

### [Additional post]( https://ultimaker.com/en/community/49572-cura-26-beta?page=6&sort=#reply-184602)
> This doesn't really look like a fair comparison. The Cura print has skin at places where S3D doesn't have that.

zerspaner-gerd, 2017-07-16:
> This doesn't really look like a fair comparison. The Cura print has skin at places where S3D doesn't have that.

I agree, there are differences!
S3D does not yet print a top / bottom layer, and Cura prints additionally a top / bottom layer!
But even these could be connected to each other, but S3D does not always do it! Or. In this part S3D does not press any.
[image]

First print the lines (No.2) then the lines (No.1) are printed, in between over 150mm travel away. (Not printed one after the other)

smartavionics, 2017-07-16:
Yes, at the moment, for each part and layer, Cura will print all of the wall outlines before printing any infill or skin. The wall outlines are printed in sequence so it will print one outline around each hole first then go back and print the second outline and so on.
A well optimised Cura would print all of the outlines and infill and skin in a given region before moving on to the next region. The program is not yet structured to be able to do that but work is underway at least to optimise the order of the wall outlines and that can remove quite a few travel movements. One step at a time...
If you are interested, the wall optimisation PR is https://github.com/Ultimaker/CuraEngine/pull/524

BagelOrb, 2017-07-21:
1. S3D has infill pattern set to Zig Zag
2. S3D is not generating skin there. There's a Cura feature on it's way to limit the amount of skin generated for slanted surfaces.
3. The order is not optimized. For each part it prints all the infill after/before all the skin. Because there are several  infill/topbottom parts enclosed in walls which are quite far apart this causes long travel moves.

In order to solve 3 we'd need to split the area within the inner walls into parts and print all the infill/skin of each of those fill parts before moving on to the next.
That means we'd have to restructure the way the areas are stored.
Instead of Layer > Parts > walls + infill + skin parts, we'd need to have Layer > Parts > walls + (fill parts > infill + skin parts).
Something like
```
class FillPart {
  Polygons infill;
  std::vector<SkinPart> skin_parts;
}
class LayerPart {
  std::vector<Polygons> insets;
  std::vector<FillPart> fill_parts;
}
```

I believe this bug occurs quite often, because I think quite a lot of people are printing covers and cases which are thin everywhere except a couple of places where there's infill/skin.

I don't know when we will have the time to fix this, though :/

zerspaner-gerd, 2017-07-22:
@BagelOrb To
> 1.  S3D has infill pattern set to Zig Zag

Would be the honeycomb infill.

> 2. S3D is not generating skin there. There's a Cura feature on it's way to limit the amount of skin generated for slanted surfaces.

What is the name of the function?
Could not find them

@smartavionics 
Is there any guidance on how to test what needs to be done (for example, integrating master twigs in Cura)

smartavionics, 2017-07-23:
> Is there any guidance on how to test what needs to be done (for example, integrating master twigs in Cura)

Sorry, I don't really understand your question, can you please rephrase it?

BagelOrb, 2017-07-23:
The feature I'm talking about is not yet merged into the master branch.

It's called Skin Reference Wall. The naming is a bit obscure, but you'll have to read teh description a couple of times ;)

closed 2018-10-16 by Ghostkeeper