Ultimaker/Cura #2123: Ironing

ChrisRiddell, 2017-07-16:
Hello,

I hope it's not to early to get this report underway, I have been playing around with the master branch and the Ironing and noticed an irregularity in the process.

**Issue one**

When there is larger movements of the nozzle it seems to start skipping lines in the ironing see eg's below. I have played around with the settings available flow, line, speed etc but the gaps in the ironing on larger movements still stayed.

eg 1:
[image]

eg 2:
[image]
https://www.thingiverse.com/thing:2358331

This is using a stock Ultimaker 2+ profile, and a ruby nozzle I'll be trying a normal brass nozzle tomorrow and will update but I'm not 100% that's the issue as this only seems to affect larger movements and not all the movements.

**Issue Two**

This is more of a really nit picky thing but the ironing speed setting should be with the rest of the ironing setting in the experimental tab and not under the speed setting until ironing comes out of experimental status.

ChrisRiddell, 2017-07-20:
Update one Issue one; Tried a brass nozzle and seen little difference.

Ghostkeeper, 2017-07-24:
I've seen that banding too, but it's not caused by skipping lines. It's actually caused by those lines taking longer to print and as a result the material doesn't cool down as much before the nozzle goes over it again. If it's a very short piece, the nozzle goes over the same area a lot of times in a short period and as a result the material sags more or degrades a bit.

You could try the concentric pattern, but that has other problems.

Ghostkeeper, 2017-07-24:
Also, if you have to do a retract, that messes up the surface a lot too.

ChrisRiddell, 2017-07-24:
ahh okay.

not sure if it's possible what about having it say if these line(s) take x amount of time speed up by a factor of x?

Ghostkeeper, 2017-07-24:
That's possible, at least technically (architecturally it's a bit harder but doable).

But would you want to speed up or slow down when you're printing short lines?

If you're slowing down, it takes longer to get back to the place (so it cools down further) but you're staying longer on every part of the print because you're going slower. If you're speeding up, it takes shorter to get back to the place (so the original problem is worsened) but you're not heating the material up as far as you would normally.

closed 2018-02-25 by ChrisRiddell