Ultimaker/Cura #2124: Feature request: save model file names in project so F5 works after project is loaded

smartavionics, 2017-07-16:
OK, so I'm just getting to grips with Cura projects and I have a project that has an infill mesh and a non-infill mesh that are loaded as separate models. It is nice to be able to reload the models using F5 but it seems that once the project is saved and reloaded, the names of the model files are forgotten and so the models can't be reloaded using F5.

So the feature request is to save the model file names in the project so that the models can be reloaded after the project has been loaded.

nallath, 2017-07-17:
Project saving is tricky with reload. 
A few cases we need to think about; 
1. Model changed in project, but the transformation didn't 
2. Mode stayed the same, transformation changed in project
3. Someone added a model to the cura buildplate.

So if one re-loads a project, should then the entire state be resetted to that project, or just specific things? If it's the second, in what cases should the change happen?

smartavionics, 2017-07-17:
Hi. I would have thought that at a minimum, being able to reload the models would be useful. I guess that means the transformations have to be saved as well in case the size/positions of the models need to be changed.

awhiemstra, 2017-07-17:
Hmm, I think what @smartavionics means here is rather than reloading the entire project, add some way to indicate the "origin" of a mesh in a project, so that later on you can reload individual meshes when they change on disk. I think this is a valid feature request, but it does get tricky when determining what to reload when you do "reload". Maybe we can distinguish per-model reload from "reload all" somehow?

smartavionics, 2017-07-17:
Yes, that's what I'm asking for, the ability to be able to reload the meshes, ideally, an individual mesh. Actually, F5 always reloads all the meshes so perhaps there's another feature request there to be able to reload only the selected meshes if any are selected rather than reloading all meshes.

fusentasticus, 2017-10-08:
This is clearly a bug, please refile as such. Indeed, Ultimaker writes "**project will load exactly as it was saved**", see [Cura Manual](https://ultimaker.com/en/resources/38954-cura-project-saving). This is exactly of course what a "project" concept is in all kinds of software tools!

Currently, Cura does not do reload correctly as OP points out. 

Thanks!!!

Ghostkeeper, 2018-02-05:
This was removed from our planning by our project manager because it's older than 12 weeks.

smartavionics, 2018-02-05:
That's great - just ignore for 3 months and the problem goes away. I wish I could solve all of my problems by doing that.

This is still required, I shall just create a new issue.

Ghostkeeper, 2018-02-27:
Fine, I'll close this one as a duplicate of https://github.com/Ultimaker/Cura/issues/3232 then.

closed 2018-02-27 by Ghostkeeper