Ultimaker/Cura #2126: Feature request: hide Dual Extrusion tab

ChrisRiddell, 2017-07-17:
It would be nice if the Dual Extrusion tab in the side menu was hidden on machines that do not have dual extrusion, Maybe have it like the Travel tab if there is no settings to show hide the whole tab.

fieldOfView, 2017-07-18:
Agreed. This should actually already be the case, so it is sort of not a feature request but a bug report.

LipuFei, 2017-07-18:
@fieldOfView I think you have fixed this in https://github.com/Ultimaker/Cura/pull/2108

fieldOfView, 2017-07-18:
No, (I think) this is about the category header in the sidebar, whereas that PR is about the extruder button in the toolbar.

The problem with the category header is that SettingDefinitionModel:_isAnyDescendantVisible() only checks if there are settings that are marked to be "visible" (in the Settings Visibility page), but not if they are "enabled" for the current stack. In this particular case, ```"Enable Prime Tower"``` is a setting that is made visible by default, but it is disabled (hidden) with ```"enabled": "machine_extruder_count > 1"```. So you get an empty category.

ChrisRiddell, 2017-07-18:
You are correct this is about the category header.

fieldOfView, 2017-07-18:
I have made a quick fix. https://github.com/Ultimaker/Uranium/issues/267

closed 2017-07-20 by nallath