Ultimaker/Cura #2130: Cura adds bottom surface to shapes that should have open underside

fuquacc, 2017-07-18:
If I make a cube with an open bottom and a rim around the inside of the bottom opening, Cura 2.5.0 closes the bottom surface of the shape.

Here's a screenshot of the shape in SketchUp:

[image]

And here it is in Cura:

[image]

If I remove the inner rim, Cura does not close the bottom. Again, here it is with no inner rim in SketchUp:

[image]

and in Cura:

[image]

Viewing the .stl files in MeshLab indicates that SketchUp is correctly exporting the shapes to stl.

The cube is 50mm on an outer edge, with 5mm-thick walls and ceiling. The inner rim is 5mm wide and 1mm thick. Attached below are the stl files I'm using:

[stl_files.zip](https://github.com/Ultimaker/Cura/files/1154473/stl_files.zip)

smartavionics, 2017-07-18:
FreeCAD says it had "flipped normals", I unflipped them and now Cura gives this:

[image]

fieldOfView, 2017-07-18:
@fuquacc: your model is not very suitable for 3d printing. The walls have no thickness (though the brim you modelled does). Slicers need "solid" models where each wall has a thickness, and everything is closed.

As a rule of thumb for sketchup: blue walls are the inside of you model, so they show there is a hole in your geometry. If you see blue on the outside of the geometry, you are doing it wrong. Another rule of thumb: SketchUp is very good at creating 3d models that are not suitable for 3d printing and give unexpected results.

fuquacc, 2017-07-19:
@fieldOfView Ah, sorry, I was unclear in my screenshots. The images from SketchUp contain cutplanes that hide the rest of the model in order to focus on the bottom centimeter of the model.. Here's a view from below of the version that doesn't have a brim, sans cutplane:

[image]

fieldOfView, 2017-07-19:
I'll reiterate: SketchUp is very good at creating 3d models that are not suitable for 3d printing and give unexpected results. 

If you use SketchUp for 3d printing, you're gonna have a bad time.

fuquacc, 2017-07-20:
If it's SketchUp's fault, then why do the meshes look correct in MeshLab? For example, here's what the stl with the inner rim looks like in Meshlab. It quite clearly has an open bottom, unlike the toolpath that Cura made for it.

[image]

This is the stl that SketchUp exported. It is identical in shape to how it appears in SketchUp (and is completely manifold, as reported by MeshLab), which implies that SketchUp is not to blame here.

fuquacc, 2017-07-20:
@smartavionics that toolpath screenshot you posted definitely looks right. Did you use FreeCAD to flip the normals?

smartavionics, 2017-07-20:
Hi, yes, that's exactly what I did, and then Cura was happy with the model.

nallath, 2017-07-20:
Because there is a major difference between looking correct and being correct. 

STL is a surface based 3d model, which is actually quite bad for 3D printing. For a 3D print you don't care all that much what the surface is, you need to know what parts need to have material and what parts don't. A single plane is a perfectly valid STL, but is not printable, as the printer can't make an infinitly thin object. If the object is not watertight, it could also cause issues, as we don't know what is "in" and "out"

fuquacc, 2017-07-20:
Right. That's what I was getting at when I pointed out that MeshLab said the mesh was manifold. I'll look into FreeCAD.

Thanks!

BagelOrb, 2017-07-21:
Don't model in Sketchup, it adds invalid geometry on the inside of your model, which f*cks up how the 3d surface can be interpreted as describing a solid model.

You should be able to see red faces in the x ray view, which show where the problem is.

If you don't want to you could always try fiddling with the settings in the mesh fix settings category.

closed 2017-07-21 by BagelOrb