Ultimaker/Cura #2132: Origin included in convex hull for models loaded as 3mf

fieldOfView, 2017-07-18:
When loading a 3mf file, the origin seems to be included in the convex hull. This can make the convex hull unnecessarily large, possibly making the model unprintable because it no longer fits on the buildplate.

[image]
NB: in the screenshot the object appears grouped. Even after ungrouping, the issue remains.

Files: https://www.thingiverse.com/thing:2440045
3mf file created with 3d builder.

Originally reported here: https://www.reddit.com/r/3Dprinting/comments/6nu2it/why_am_i_getting_this_shadow_tail_when_i_try_to/

nallath, 2017-07-19:
It doesn't happen on master as far as I can see. The grouping there is a bit off, but i don't get the weird elongated shadows.

fieldOfView, 2017-07-19:
It does for me. But I could be using an older libSavitar.

nallath, 2017-07-19:
I removed the other objects and put that one in the center.

I did change something in lib savitar that would always put (0,0,0) verts in the model. So that should have some impact.

fieldOfView, 2017-07-19:
> I did change something in lib savitar that would always put (0,0,0) verts in the model.

That sounds exactly like what it looks like for me, so I guess it is fixed. Yay!

fieldOfView, 2017-07-19:
> I did change something in lib savitar that would always put (0,0,0) verts in the model.

Have you pushed those changes? I don't see anything like that in master (compared to 2.6), and I just updated my libsavitar to master and this is what I see when loading the 3mf file:
[image]

nallath, 2017-07-20:
Yep, just checked. I've attached a screenshot how it looks for me. 
[image]

The issue was that I used a reserve on a vector (which would fill it up with 0,0,0) vectors and then started appending to the back. So for every vert in the file it would add an extra point on the origin.

ianpaschal, 2018-07-30:
I'm not able to reproduce this anymore on Cura 3.4.1. I'm pretty sure it's fixed.

closed 2018-07-30 by ianpaschal