Ultimaker/Cura #2133: Filament diameter doesn't change

Eternyt, 2017-07-18:
I set up a new printer with a filament diameter of 1.75, but in the default Cura profiles, the diameter remains at 2.85. No way to change it permanently, also if I remove the printer and create a new one the diameter doesn't change from 2.85.

fieldOfView, 2017-07-18:
If you change your material diameter in the "custom" setting in the sidebar, it will stick. If the material diameter is not visible in the sidebar, you can just type in "diameter" in the search bar and it will show up. Put "1.75" in there.

This was recently fixed for Cura 2.7.

Eternyt, 2017-07-18:
Yes, I can change it, but not permanently because I can't update the default Cura profile. I'm happy to hear that in Cura 2.7 this problem will be fixed. Thanks!

fieldOfView, 2017-07-18:
> Yes, I can change it, but not permanently because I can't update the default Cura profile.

That should not really matter. Once you enter "1.75", you should be switched to 1.75mm materials. Perhaps you have to (re-)select a material.

The problem in 2.6 is that the diameter set for the machine is always overridden by the material, and the chosen material is 2.85mm by default.

Eternyt, 2017-07-18:
Got it! Thanks!

closed 2017-07-18 by Eternyt