Ultimaker/Cura #2134: Linux 64 bits issue

mariolpantunes, 2017-07-18:
Hi, in order to compile Cura (and all the libs) I had to patch up the CMakeLists.txt to correctly use Lib_Suffix.
I share my lib64.patch, if you found it acceptable can you add it to the code?
[lib64.txt](https://github.com/Ultimaker/Cura/files/1156564/lib64.txt)

Ghostkeeper, 2017-10-28:
This should now be fixed, albeit slightly differently: https://github.com/Ultimaker/Cura/commit/6484df3e60d194e0d31f7704865b552e50423591

Could you try if it works?

closed 2017-10-28 by Ghostkeeper

mariolpantunes, 2017-11-30:
Will you make a release that has this fix? Or should I pull the git?

Ghostkeeper, 2017-12-13:
I'm not sure what you mean. We've made releases: https://github.com/Ultimaker/Cura/releases

But those are binary builds. We've already built them, so this fix for building Cura on non-Debian systems won't have any effect since they are built correctly on our CentOS build server.

So yes, you'll have to pull the git. If you want a stable version you can pull the 3.1 tags on all of Cura's repositories.