Ultimaker/Cura #2135: Connecting to Non Ultimaker Printers

meldrex5, 2017-07-18:
If Cura is the first program opened after connecting a printer it will automatically connect to it.  This alone is annoying but then there is no option to disconnect from it, you have to close Cura, connect with the other interface then reopen Cura.

Ghostkeeper, 2017-07-24:
Also happens with the Ultimaker Original.

astecme, 2017-07-26:
Indeed. Cura does not return to the main program after finishing and the only way out is to shut Cura down.

Ghostkeeper, 2017-08-07:
Oh, well you should be able to switch back to the settings view while it's printing... Aren't you?

Ghostkeeper, 2018-02-05:
Our project manager just removed this from our planning because it's older than 12 weeks.

rlesac, 2018-11-21:
It would be lovely to be able to disconnect from the printer without shutting down Cura. 
Please reconsider.

metaxis, 2018-11-21:
an option to manually dis/connect vs autoconnect would be great.

yodor, 2018-11-22:
I keep pushing the reset button of my printer to help Cura recognize it. Would be nice to have port/speed/connect button panel.