Ultimaker/Cura #2138: Connect to several Octoprint with different API key

UM2EXTENDED, 2017-07-20:
Cura 2.6.2 OS X
Detected Octoprints
octopi1._octoprint._tcp.local
octopi1._octoprint._tcp.local
octopi1._octoprint._tcp.local

I have tried to Octoprint and it works but with some problems:

nallath, 2017-07-20:
@fieldOfView I can't assign you, but this is your cup 'o tea!

fieldOfView, 2017-07-20:
The OctoPrintPlugin is maintained here: https://github.com/fieldOfView/OctoPrintPlugin

Handling multiple octoprint instances with different API keys more clearly is on my list for Cura 2.7:
https://github.com/fieldOfView/OctoPrintPlugin/issues/46
Note that it is possible to have a different API key for each octoprint, though it is not very intuitive. The api key is stored/used when you press connect. 

The camera orientation has already been implemented for Cura 2.7:
https://github.com/fieldOfView/OctoPrintPlugin/issues/29

ChrisTerBeke, 2018-03-02:
Closing here as it's a 3rd party plugin issue.

closed 2018-03-02 by ChrisTerBeke